bigtreetech/BIGTREETECH-TouchScreenFirmware #1383: [BUG] Click to Resume on startup in Marlin Mode

blueeagle69, 2020-12-13:
Hi.

I just flashed the latest Marlin bugfix.
When the printer restarted in Marlin mode I had "Click to Resume" on the bottom of the screen.
I clicked the encoder and it appeared again. After clicking it a second time it goes away.
The reason I posted here and not the Marlin repo is I have had this before with the TFT.
If I go into TFT mode and reset the TFT to default settings, the Click to Resume on printer start goes away when in Marlin mode.
But weirdly, if I do reset the TFT using Menu/Settings/Feature/Reset default settings, my Show Title I set to 0 gets wiped out and reset back to 1.
But the Click to Resume in Marlin mode goes away.
Apparently it's caused by a buffer overflow of some sort?

blueeagle69, 2020-12-13:
Just flashed the latest Master. I thought it was fixed.
But on the third boot-up it returned  🙄
[image]

blueeagle69, 2020-12-13:
Yet another update.
I re-edited the TFT config.ini and disabled serial_always_on (set it back to 0).
I have restarted my printer 5 or 6 times in Marlin mode and the "Click to Resume" has gone.
Looks like the Serial Always On being enabled is acting up again?

radek8, 2020-12-13:
I had similar problems. After restarting the printer, Marlin stopped and waited for confirmation, sometimes twice in a row.
I found out it makes an M1 command that Marlin accepts.
Unfortunately, I did not find in the TFT code that would send this command. By analyzing the packets sent by the TFT, I found that immediately after turning on and connecting the TFT, it repeatedly sends the M105 command, which was damaged due to interference, and the printer accepted it as M10105, M10M105, 05, M1005, MM105, or M1.
It was the last named M1 that caused the printer to stop.
Look at the screanshot in # 1297 to see the printer's communication with TFT.
During the printing, the interference didn't show much to me, only occasionally the printing stopped and if I wasn't there to confirm the pause, the printing was destroyed.
Conversely, for some users, interference starts to appear after turning on the bed heating and reaching the set temperature, when PWM control starts to apply.
I solved the problem with a new shielded cable on the serial port.
The cable supplied by BTT is very bad and unsuitable and often causes problems.

[image]

radek8, 2020-12-13:
> Yet another update.
> On a hunch, I re-edited the TFT config.ini and disabled serial_always_on (set it back to 0).
> I have restarted my printer 5 or 6 times in Marlin mode and the "Click to Resume" has gone.
> Looks like the Serial Always On being enabled is acting up again?

If you set serial_always_on 0 and restart in Marlin mode, the M105 command will not be sent after startup and the printer will not stop.
After switching to touch mode, the interference is no longer so strong and the display works.
Has it ever happened to you that when printing from TFT, the tick stopped in the middle of printing and destroyed the printing?

blueeagle69, 2020-12-13:
Hi.

No, not yet!

Bastlwastl84, 2020-12-14:
> I had similar problems. After restarting the printer, Marlin stopped and waited for confirmation, sometimes twice in a row.
> I found out it makes an M1 command that Marlin accepts.
> Unfortunately, I did not find in the TFT code that would send this command. By analyzing the packets sent by the TFT, I found that immediately after turning on and connecting the TFT, it repeatedly sends the M105 command, which was damaged due to interference, and the printer accepted it as M10105, M10M105, 05, M1005, MM105, or M1.
> It was the last named M1 that caused the printer to stop.
> Look at the screanshot in #1297 to see the printer's communication with TFT.
> During the printing, the interference didn't show much to me, only occasionally the printing stopped and if I wasn't there to confirm the pause, the printing was destroyed.
> Conversely, for some users, interference starts to appear after turning on the bed heating and reaching the set temperature, when PWM control starts to apply.
> I solved the problem with a new shielded cable on the serial port.
> The cable supplied by BTT is very bad and unsuitable and often causes problems.
> 
> [image]

Do you have a link to this shielded cable?

oldman4U, 2020-12-14:
Maybe your ticket #1199 is somehow related?

radek8, 2020-12-14:
I do not have a download link, here is the production instructions :-)

[image]

blueeagle69, 2020-12-14:
> Maybe your ticket #1199 is somehow related?

I'm not sure TBH.
I just noticed that the Marlin LCD mode started chucking out "Click to continue"  messages when the printer was switched on, and for now ate least fixed it by disabling Serial Always On 😕

oldman4U, 2020-12-16:
As far as I understand this function is more for keeping the serial connection on also while using Marlin emulation. Serial has to be on while using Touchscreen, otherwise it will not work at all;-)

And with Marlin 2.0.7.2 or an earlier version, this problem does not exist?

blueeagle69, 2020-12-19:
Hi.

For now I have just disabled that feature and printed from Marlin mode.
Seems to work fine with no Click to resume constantly popping up on boot 😀

blueeagle69, 2020-12-21:
Do you want me to close this @oldman4U  😀

oldman4U, 2020-12-21:
I can not see, how this can be a firmware issue and even users like radek8 help in such situations, it would be good to close tickets which are not really related;-)

Happy holidays

blueeagle69, 2020-12-21:
Merry Christmas all!

PR #1468 (2021-01-07): save/load infoSettings Struct instead of save/load every variable individually, Fix Screen freezes/Click to Resume message/damaged M105 command in Marlin mode
- The settings are stored in the last 2KB of the program memory. earlier a separate entry was used to save/ load each variable. With this the **infoSettings** Struct is copied at once to/from buffer instead of saving/loading every variable individually, removing the need to add save/load entries for each variable. This also eliminates bug related to wrong/invalid settings being loaded. The prebuild settings size check now works accuratel [... cut]
closed 2020-12-21 by blueeagle69