bigtreetech/BIGTREETECH-TouchScreenFirmware #1385: [Q] (BIGTREETECH BTT PITFT50 support brightness control?)

savenko-egor, 2020-12-13:
**Your question or details you need. Please describe.**
I got my PITFT50 display and try control brightness via [rpi-backlight](https://pypi.org/project/rpi-backlight/). I tried from pi user and from root but don't got any effect. Please specify I can control brightness with my PITFT50?

**specify the device variant related to your question.**
[BIGTREETECH BTT PITFT50](https://aliexpress.com/item/1005001270623489.html)

Thakns!

oldman4U, 2020-12-13:
Hi.

This repository is related to the FIRMWARE for the BTT TFT touchscreens. Have you tried to ask your question in a RPi user group?

savenko-egor, 2020-12-13:
This question about display supported control brightness or not. Only this.

oldman4U, 2020-12-14:
So you know this is the wrong repository but hope that someone can answer the question?

closed 2020-12-14 by savenko-egor

bigtreetech, 2020-12-15:
Hello, PITFT50 don't support brightness control.