bigtreetech/BIGTREETECH-TouchScreenFirmware #1386: [BUG] Onboard SD print - Layer display stops at x height

kisslorand, 2020-12-13:
### Description

Printing a small object from onboard SD and the layer height stops at 0.80 mm. If I do nothing it stays that way, if I click "More" and than "Back" the layer height is updated until the end of the print.

### Steps to reproduce

1. [Start print]
2. [Watch layer height]
3. [and so on...]

**Expected behavior**

 Layer to update until the end of the print job.
**Actual behavior**

Layer stops at 0.8mm
### Hardware Variant

Clone of the MKS TFT 28

### TFT Firmware Version & Main Board Firmware details

Latest build as of 14.XII.2020
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
Here's the gCode:

[Strap6mm.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5685793/Strap6mm.zip)

oldman4U, 2020-12-14:
Like always, it works on my printer. In addition your gcode file surprises the print finished dialog and uses the classic print finished method.
[Layer_s.mov.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5688041/Layer_s.mov.zip)

kisslorand, 2020-12-14:
I got it. It's a programming error. There was a ternary in an "if" conditional. It might work or it might not. In my case it didn't.

Instead of:
`if (a != (b=c) ? d : e)`
`___doSomething();`
one should use:
`f = (b=c) ? d : e;`
`if (a != f)`
`__doSomething();`

Grrrr, it took me 5-6 hours to find it...

The logical value of 0.xx is FALSE. As soon as the Z hit 1.00 it's logical value changed to TRUE. Using ternary in the "if" conditional made the layer comparisons to be logical comparisons, not numerical ones. In logical terms 3.30 equals 8.60, in numerical terms they are not equal.

oldman4U, 2020-12-14:
Where can I find: if (a != (b == c) ? d : e)

Seems it does not exist in my VSC.

kisslorand, 2020-12-14:
It's in "PrintinMenu.c", right under the "//Z_AXIS coordinate" comment.

guruathwal, 2020-12-15:
> I got it. It's a programming error. There was a ternary in an "if" conditional. It might work or it might not. In my case it didn't.
> 
> Instead of:
> `if (a != (b == c) ? d : e)`
> `___doSomething();`
> one should use:
> `if (a != ((b == c) ? d : e))`
> `___doSomething();`
> 
> Grrrr, it took me 5-6 hours to find it...

This is a bit strange. How can a logic change its behaviour by just switching the menus? both are completely independent of each other.

kisslorand, 2020-12-15:
When "curLayer" becomes "1.00" the "curLayer != (infoFile.source == BOARD_SD)" becomes FALSE.
When switching to another menu and than back "curLayer" gets initialized again and it's value becomes "0" and after that it will be higher than "1.00" so "curLayer != (infoFile.source == BOARD_SD)" will be TRUE.
"curLayer != (infoFile.source == BOARD_SD)" is FALSE only when "curLayer" is "1.00".

guruathwal, 2020-12-15:
> When "curLayer" becomes "1.00" the "curLayer != (infoFile.source == BOARD_SD)" becomes FALSE.
> When switching to another menu and than back "curLayer" gets initialized again and it's value becomes [...]

yes. 👌

oldman4U, 2020-12-15:
Changed it and asked my son to see if the layer high updates. Even it has been working fine before;-)

When will you add this to your PR?

kisslorand, 2020-12-15:
It might be tomorrow or the day after tomorrow.  I will integrate it in PR #1359 .  If anyone wants so to make a PR with this fix sooner, I don't mind it.
BTW, Marlin just merged my PR about this bug, so we're good to go with the onboard SD printing as it was meant to be.

oldman4U, 2020-12-15:
No need to hurry I assume.

So Marlin update is needed when printing from SD Card. Maybe it would be
good to update the information about the dependency in config.ini also?

kisslorand, 2020-12-15:
A Marlin update is not necessary but only if you want the print finish to be as it was designed.
Without a Marlin update the print itself will be just as before.

oldman4U, 2020-12-15:
I would update the Marlin information. 

Great work!

kisslorand, 2020-12-15:
Well... thanks for the appreciation but it was me who introduced this bug (or so I think).

oldman4U, 2020-12-15:
You solved it and it was there because you coded a great feature.

digant73, 2020-12-16:
@kisslorand 
is the fix simply:
if (a != ((b == c) ? d : e))
in that case I could add it in my current minor PR (just to speedup the merge). Let me know if I can add it

kisslorand, 2020-12-16:
Yes, that's THE fix. Please add it to your PR. Thx

oldman4U, 2020-12-16:
Thank you.

Do you want to close this ticket;-))))

kisslorand, 2020-12-16:
Fixed. Closed.

PR #1394 (2020-12-19): minor cleanup
just a minor cleanup (not committed on my just merged PR)

fixes #1386
closed 2020-12-16 by kisslorand