bigtreetech/BIGTREETECH-TouchScreenFirmware #1388: Icons, text and menus scrambled TFT28

JRosello, 2020-12-14:
My TFT 28 is driving me crazy... since I first updated the firmware to use a new theme, some parts are messed up.
Broken and missing icons, weird text (characters) on top the Marlin simulator mode, etc...

I read past issues about that I tried to fix it with no results...
-Reformatting the SD to FAT32
-Every time I try to reinstall in the SD card are all items:
TFT28 folder with font and bmp foders inside
The firmware, in my case BIGTREE_TFT28_V3.0.26.x.bin
The config.ini file
And the language language_en.ini

In all the cases the update process show that are some missing bmp files, but I checked and they are in the bmp folder.
My modified 320x240 24bit bmp logo.bmp file never loads properly. Always looks like a corrupted files with weird patterns.

I always use the pre-compiled .bin file in the "BIGTREETECH-TouchScreenFirmware-master\Copy to SD Card root directory to update\BIGTREE_TFT28_V3.0.26.x.bin", never compiled my self the firmware.

So anyone can bring any light?
Thanks

oldman4U, 2020-12-14:
Hi. What is the firmware version shown under Info?

JRosello, 2020-12-15:
Marlin firmware is the latest bug fix release 2.0.8
The board model is BIGTREETECH_TFT28_V3.0
The Board Firmware is TFT28_V3.0.26.x (latest release I downloaded yesterday)

JRosello, 2020-12-15:
[image]

oldman4U, 2020-12-15:
OK.

Unfortunately I have no v3 but only a v1 TFT28, but I have been testing the precompiled firmware using the v1, also with config.ini and en language as well as TFT28 folder and it worked fine.

What is the size of the SD card you use to update the firmware?

JRosello, 2020-12-15:
4Gb Sd FAT32

oldman4U, 2020-12-15:
Could you make a movie which shows the update process and also a screenshot of the content of the SD Card before and after update.

Thank you

JRosello, 2020-12-15:
SD Card contents... let some minutes and I tro upload a video...
[image]
[image]
[image]
[image]
[image]

oldman4U, 2020-12-15:
The files seem to be fine and what you see on screen is definitely wrong.

Have you ever tried to reset the TFT according to the ReadMe which you can find on the first page of this repo?

JRosello, 2020-12-15:
Now it's worst.... I cannot even update the firmware... starts the process but no update is made... the contents on the sd card didn't change to .CUR ....
Maybe it's a faulty unit?
The video: https://drive.google.com/file/d/1yNWHf5mRTq770gtKX8ElFyq2jo5XpAIG/view?usp=sharing

JRosello, 2020-12-15:
> 
> 
> The files seem to be fine and what you see on screen is definitely wrong.
> 
> Have you ever tried to reset the TFT according to the ReadMe which you can find on the first page of this repo?

Reset? where? I cannot find how... I only see the reset the touch screen using a blank file called reset... but it's for the touch calibration options right?

oldman4U, 2020-12-15:
No access to your GoogleDrive...

Most likely the issue is the SD Card. Try to low format the card or use another one and try it again.

JRosello, 2020-12-15:
Sorry... try again now...
https://drive.google.com/file/d/1yNWHf5mRTq770gtKX8ElFyq2jo5XpAIG/view?usp=sharing

The reset procedures says this...
_To reset the TFT's touch screen calibration, create a blank file named reset.txt and place in root folder of the sd card. Insert the SD card into the TFT's SD card reader and power cycle your printer to start the reset process_

I'll try to low level format the SD

oldman4U, 2020-12-15:
Can you show the whole update incl. TFT folder and config.ini.

Exactly

OK

JRosello, 2020-12-15:
The whole config.ini

#--------------------------------------------------------------------
#
# DEFAULT CONFIG FILE FOR BigTreeTech TFT CONTROLLERS
#
#--------------------------------------------------------------------

#--------------------------------------------------------------------
# Supported TFT Variants
#--------------------------------------------------------------------
#
# BIGTREE_TFT35_V1_1 / BIGTREE_TFT35_V1_2    / BIGTREE_TFT35_V2_0
# BIGTREE_TFT35_V3_0 / BIGTREE_TFT35_E3_V3_0 / BIGTREE_TFT28_V1_0
# BIGTREE_TFT28_V3_0 / BIGTREE_TFT24_V1_1    / MKS_32_V1_4
# MKS_28_V1_0
#
# Firmware source: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
#

#--------------------------------------------------------------------
# Supported Marlin Firmware Versions
#--------------------------------------------------------------------
#
# Minimum Marlin firmware version: 2.0.5.4
# Distribution date: 2020-05-12
# Firmware source: https://github.com/MarlinFirmware/Marlin/releases
#
# IMPORTANT NOTE:
#   Ensure that the following options are enabled in Marlin firmware
#
# General options:
#   M115_GEOMETRY_REPORT (in Configuration_adv.h)
#   M114_DETAIL (in Configuration_adv.h)
#   REPORT_FAN_CHANGE (in Configuration_adv.h)
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#
# Options to support printing from onboard SD:
#   SDSUPPORT (in Configuration.h)
#   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
#   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)
#
# Options to support (Un)Load menu:
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)
#

#--------------------------------------------------------------------
# General Settings
#--------------------------------------------------------------------
#

#### Unified Menu / Classic Menu
# Select a UI menu flavour
#
# Options: [Unified Menu: 1, Classic Menu: 0]
#
unified_menu:1

#### Baudrate
# Options: [2400: 0, 9600: 1, 19200: 2, 38400: 3, 57600: 4, 115200: 5, 250000: 6, 500000: 7, 1000000: 8]
#
baudrate:5

#### Default Touch Mode Language
# Select the language to display on the LCD while in Touch Mode.
# To add/flash a second language copy the required "language_xx.ini" file from "Language Packs" folder
[... 553 lines cut ...]
#### Default Cancel Gcode Status
# Options: [enable: 1, disable: 0]
#
cancel_gcode_enabled:0

#### Start Gcode
# Run this gcode before starting print
#
# Value range: [max 75 characters]
#
start_gcode:G28 XY R20\n

#### End Gcode
# Run this gcode after finishing print
#
# Value range: [max 75 characters]
#
end_gcode:M104 S0\nM140 S0\nM18\n

#### Cancel Gcode
# Run this gcode after canceling print
#
# Value range: [max 75 characters]
#
cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\n

oldman4U, 2020-12-15:
> order it matches the actual axis Format: [X Y Z E] Options: [enable: 1,
> Bed / Extruder / Chamber Maximum Temperatures Format: [max_temp: T0: T1:
> T2: T3: T4: T5: BED: CHAMBER:] Unit: [°C]
> Fan Maximum PWM Speed Format: [fan_max: F0: F1: F2: F3: F4: F5: CtL: CtI:] Value
> Configuration_adv.h in Marlin firmware) Format: [size_min: X Y Z] [size_max:
> X Y Z] Unit: [mm] Value range: [min: -2000, max: 2000] Options: [X axis:
> [pause_retract: R P] [pause_pos: X Y] [pause_z_raise: Z] [pause_feedrate:
> feed rate Format: [level_edge_distance: ] [level_z_pos: Z] [level_z_raise:
> Z] [level_feedrate: X<feedrate mm/min> Y<feedrate mm/min> Z<feedrate
> Preheat Temperatures Format: [T B] Unit: [°C] Value range: [filament

Not the files but a movie which shows the update.
Sorry

JRosello, 2020-12-15:
> 
> Can you show the whole update incl. TFT folder and config.ini.
> 
The TFT\bmp\ has 136 .bmp files and one Readme.md

This is the readme.md file contents:
```
ImageMagick command

convert Reserve_6.bmp -alpha set -define bmp:format=bmp3 -define bmp3:alpha=true Reserve_6.bmp
```

JRosello, 2020-12-15:
The failed update video:
https://youtu.be/CpkEdpESgXE

oldman4U, 2020-12-15:
Please make a movie like above but where everything is updated. Firmware, config.ini and TFT folder.

JRosello, 2020-12-15:
> 
> 
> Please make a movie like above but where everything is updated. Firmware, config.ini and TFT folder.

I'll try... because seems now I cannot update any more.... it reads the card the gets the 100% firmware read and then jumps straight to the main menu... no update :/
I'll try another sd... I don't know if I have any SD below 8gb though

JRosello, 2020-12-15:
@oldman4U Wow I just used a microSD with an adapter and updated successfully!!
is this normal?? the branded Kingston SD 4Gb didn't work an a Chinese not branded cheep microsd works?!

oldman4U, 2020-12-15:
The reset.txt resets everything incl. the calibration. Try it.

oldman4U, 2020-12-15:
What is normal...

I have more than 10 identical 4GB sd cards and one I can not use for
firmware update but works flawlessly otherwise;-)

Glad it works!!!

Please help the community and close the ticket once you do not need it
anymore. Thank you

JRosello, 2020-12-15:
I'll keep in mind the reset.txt file just in case
Thanks for all!!

closed 2020-12-15 by JRosello

oldman4U, 2020-12-15:
You are welcome 

Happy printing