bigtreetech/BIGTREETECH-TouchScreenFirmware #1392: BTT TFTv2.0 touch buttons don't operate printer, connected fine

Daparat, 2020-12-15:
BTT TFT v2.0 connected on serial to SKR 1.4 Turbo, tft reports temperature changes when printing but all buttons are unresponsive, i.e press home x nothing happens, go to repetier press home x , x homes. Press nozzle temp 50, nothing happens, repetier nozzle 50 nozzle heats to 50 and displays 50/50 on tft. How can I get the icons to send/do the functio non the printer,

thanks

oldman4U, 2020-12-15:
So you can navigate to Menu - Move - Home, but then in the Home menu the buttons are not working anymore?

Has it ever been working before?
Have you tried to reset the TFT according to the ReadMe at the front page of this repository?

Daparat, 2020-12-16:
Yes the buttons work fine for navigation through the menus, status OK, reports temps correctly but when I press any button that would send any type of control to the printer it just beeps and does nothing. I've put an old MKS tft 35 and a robin nano in, both work fine, sending tft back, closing.

closed 2020-12-16 by Daparat

oldman4U, 2020-12-16:
Most likely the TX part of the cable is broken.

Happy printing