bigtreetech/BIGTREETECH-TouchScreenFirmware #1395: [BUG] (short description)

iBims1Schnubbert, 2020-12-15:
Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the BigTreeTech Facebook group: https://www.facebook.com/groups/505736576548648 or BigTreeTech Subreddit: https://www.reddit.com/r/BIGTREETECH/.
-->
### Description
The Firmware Update for TFT Freez after Update in selection Screen for Touch or Marlin mode

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2020-12-15:
Please update the title of this ticket. 

Attach a movie which shows the whole update process and the issue.

iBims1Schnubbert, 2020-12-15:
How do i update the title?

oldman4U, 2020-12-15:
And let me know about the sd card size and format.

iBims1Schnubbert, 2020-12-15:
check

radek8, 2020-12-15:
Check that the frame does not press on the edge of the display (remove the display from the frame and try it)
If this does not help, calibrate the touch layer with the reset.txt file

iBims1Schnubbert, 2020-12-16:
how do I do that with the reset? Copy an empty reset.txt file to the sd and boot into the display? or do you have to have another file on it?

oldman4U, 2020-12-16:
Exactly as you wrote. Alone or with other files, it does not matter.

iBims1Schnubbert, 2020-12-16:
THX worked great

oldman4U, 2020-12-16:
Whats about the ability to switch and which firmware is installed now?

iBims1Schnubbert, 2020-12-16:
Version 1.1.26 and witching is also possible

oldman4U, 2020-12-16:
Sounds great. Check the Info of your TFT and let us know firmware date.

iBims1Schnubbert, 2020-12-16:
Dec 15 2020 is the Date

oldman4U, 2020-12-16:
Perfect!

Please close the ticket once you do not need it anymore.

Happy printing

closed 2020-12-16 by iBims1Schnubbert