bigtreetech/BIGTREETECH-TouchScreenFirmware #1396: BTT-TFT35-E3-V3.0 filament runout rutine don't beep any alarm only shows a message...

mpcolman, 2020-12-15:
When filament out, TFT activate "filament runout" and appears the alert message but without any beep sound like alarm, need to be seted an alarm beep lik in Marlin mode, I have connected the sensor directly to the TFT screen.
The main drawback is that by not having some type of audible alarm, many hours can pass until you become aware of the situation yourself, wasting valuable time and wasted use of the printer.

oldman4U, 2020-12-16:
Hi. I can to verify this, but added this feature to the FR list.

Hope this helps to attract a developer or another user to develop this or reply with more information.

Happy printing

oldman4U, 2020-12-21:
Could you please attach a picture of the alert message.

Thank you

mpcolman, 2020-12-22:
@oldman4U here is... regards!
[image]

kisslorand, 2020-12-23:
It can be fixed.
What kind of beep would be suitable here? One kind of beep than silence? Multiple beeps than silence? Periodical beep(s) until OK is pressed? If periodical than at what time interval?

oldman4U, 2020-12-23:
Rule number one. Don't make other users cry even it is because of happiness. 🤪

mpcolman, 2020-12-23:
maybe could be a select menu in the config.ini ? if not  the best option I think it's Periodical 0.5 secs beep(s) until OK is pressed  with 20 secs interval , if someone it's sleepeng don't going tu jump over the bed..... thanks for the support!

kisslorand, 2020-12-23:
Things got complicated, working on this I realized there is just a short message about runout if the sensor is connected to the motherboard.
When the runout sensor is connected to the motherboard (and M75 is issued at print start), it's Marlin, who takes care what happens if runout detected. I want to have the same message and same repeated beep until acknowledged in that case also., but without the actions the TFT does when runout is detected by it (TFT), not want to mix Marlin actions with the TFT actions.

kisslorand, 2020-12-23:
Got a question. 
When runout sensor is connected to the mainboard, Marlin takes the matter into his hands. It will start the unload procedure and wait for new filament insertion. In the process several popups appear, so the initial "FilamentRunout T0" disappears without user intervention, other popup tales its place (Nozzle Parked, etc). So there is no way to know if the user is aware or not that the filament ran out.
What would be in your opinion the point, the trigger to stop the periodically repeated  beeps when runout sensor is connected to the mainboard? 
I am almost done but got this dilemma...

kisslorand, 2020-12-23:
Done. #1428

oldman4U, 2020-12-24:
You almost destroyed it🤩

mpcolman, 2020-12-24:
@kisslorand the filament out screen have an OK confirmation button, once confirmed by the user the beeps must shut off and going with the next step procedure.....thats perfectly fine for me..... thanks!

kisslorand, 2020-12-24:
That's exactly how it's done.

mpcolman, 2020-12-24:
@kisslorand i see that you're updated some files with this issue, it's ready to test it? what files need to be updated? regards!

kisslorand, 2020-12-24:
PR #1428 is included in the latest Master version now but that master has a bug that  is fixed already with PR #1431
You can compile it yourself after download it here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/bc67d1218d56f3475f66a8e115786b49d40dcd95

mpcolman, 2020-12-25:
the files to be compiled are "language Packs" directory and the config.ini right?

kisslorand, 2020-12-25:
New master is out. Check that one.

mpcolman, 2020-12-25:
sorry... Im a little lost......  i see the same updates...  "language Packs" directory and the config.ini Im right?

kisslorand, 2020-12-25:
Just download(clone/sync) the whole repository and compile it.

oldman4U, 2020-12-25:
Update is always 3 steps and 3 or 4 "elements" like described here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#update-tft-firmware

mpcolman, 2020-12-26:
updated now... but no beeps when filament runout message appears......... sorry :(

kisslorand, 2020-12-26:
Do you have filament runout sensor enabled in parameters settings?

mpcolman, 2020-12-26:
yes of course, when activate the sensor it appears the runout message.... but no beeps...

kisslorand, 2020-12-26:
What if you wait a little? It should be an acoustic warning every 10 seconds.

mpcolman, 2020-12-26:
nothing.... wait for more than a minute......

kisslorand, 2020-12-26:
Do you have sound when you press buttons an click around in the menu?

mpcolman, 2020-12-28:
yes, I'm thinking a possibilty... IE: when print are finished the info screen appears with a little beep sound, maybe you can copy the same code to apply to the filament runou screem.....

kisslorand, 2020-12-28:
What does the filament runout popup says, "filament runout!" or "Filament runout"? Starts with big "F" or a small "f"?

mpcolman, 2020-12-29:
"filament runout!"  all small.....

kisslorand, 2020-12-29:
In the version where the periodic beeps are implemented the text was changed to "Filament runout !", with big "F", so it seems you are not using the latest FW.
Check [this](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5748788/BIGTREE_TFT35_V3.0_E3.26.x.zip) firmware I uploaded for some other user but with the same TFT as yours.

mpcolman, 2020-12-31:
AWESOME!!!! now it works like a charm!!! excelent job my friend very usefull!!!! lot of thanks!!

mpcolman, 2020-12-31:
Where is the code part where is located the parameter for the sound and volume level if exist???  regards...!

kisslorand, 2020-12-31:
In extend.c:

 - `#define ALARM_REMINDER_TIME 10000` - sets the period in ms
 - `BUZZER_PLAY(sound_error)` - sets the alarm sound type

There's no possibility to change the volume.

mpcolman, 2020-12-31:
@kisslorand  great job thank you very much!

kisslorand, 2020-12-31:
It was your idea in the first place and it was a really good one. I like it very much this way.

oldman4U, 2020-12-31:
GREAT!!!

How can other users know where to change the parameters?

In extend.c:
#define ALARM_REMINDER_TIME 10000 - sets the period in ms
BUZZER_PLAY(sound_error) - sets the alarm sound type

oldman4U, 2020-12-31:
@mpcolman 

You can close this ticket at any time.

Happy printing

kisslorand, 2020-12-31:
@oldman4U 

> GREAT!!!
> 
> How can other users know where to change the parameters?
> 
> In extend.c:
> #define ALARM_REMINDER_TIME 10000 - sets the period in ms
> BUZZER_PLAY(sound_error) - sets the alarm sound type

It was never meant to be user adjustable. I was asked where are things in the code and I responded to that.

@mpcolman 
It's your ticket, you are supposed to close it

oldman4U, 2021-01-01:
Ok

mpcolman, 2021-01-01:
Thanks to all fot the support.

closed 2021-01-01 by mpcolman