bigtreetech/BIGTREETECH-TouchScreenFirmware #1400: [FR] RRF compatibility

bttguy, 2020-12-16:
**Describe the solution you'd like**
Some (and a growing number) of us use the BTT TFT for basic monitoring and controlling their SKR LPC/STM port based RRF. It would be nice to be able to monitor the print status, modify some of the parameters during printing, accessing macros, and use some of the extra features if possible like LED changing, and probing...

**Describe alternatives you've considered**
There are some forks and forked forks that try to modify the base firmware to enable as much functionality as possible, but lack the constant updates, and work as is they were made.

**Additional context**
The RRF project bases mainly on BTT boards, and there are already sollutions to make them work with BTT TFTs, so it would be nice to support it “officially”. I dont have to skill to do this and dont know how hard it would be to make and manage, but this TFT github community is quite nice as I followed during my Marlin use, so maybe some of you have the skill and time to modify and keep updating the Gcode and communication related differences in the firmware as an alternative for the users, so more of use could use this TFT in our printers. 

The project: https://github.com/gloomyandy/RepRapFirmware/wiki

Does this idea has any potential to be implemented in the future?
Thank you in advance.

radek8, 2020-12-16:
look at #1399

bttguy, 2020-12-16:
A saw it, thats why I try to push RRF compatibility even further. If you add that Gcode line, that only allow to communicate to the board at te TFT port. Would like to resolve if possible his/our issue "Due to the lack of \n when the Reprapfirmware responds to M20, it cannot read the onboard sd card normally.
Some advanced settings of marlin different from reprap are not available."
I am thinking of a paralell "official" branch that keep up with the changes and modify the code to be able to properly communicate with RRF.
This is a fork which is try to modify the FW for RRF: https://github.com/Thro42/BTT-SOVOL-TouchScreenFirmware-Duet

oldman4U, 2020-12-17:
Add @Subcircuit because he is the one who contributed related to RRF.

oldman4U, 2020-12-21:
The interest seems to be somehow limited. bttguy, are you willing to take over all the administration part and what will you do to attract developers to do the development? Or are you a developer and how can this community help?

Thro42, 2021-01-04:
If you like to bring in my expirients from https://github.com/Thro42/BTT-SOVOL-TouchScreenFirmware-Duet. For me it is better to start here from scratch. The fork from JimmymaDdness and bcmob is too far away from the actuallity one. I have to start over anyway.

bttguy, 2021-01-05:
@Thro42 This would be awesome! @oldman4U can help you how you can implement the compatibility without conflicting the main branch. Sadly I cannot code or have deeper understanding of the firmware, but can test the further changes and give you some feedback/ideas as a RRF user with a BTT TFT!

Thro42, 2021-01-06:
@oldman4U how to get a contributor?

bttguy, 2021-01-08:
@radek8 @oldman4U @digant73 @bigtreetech  @thisiskeithb (I see your names a lot, so thanks for the work, and I trust you with my question :) )
Can you give us some infos about how this RRF compatibility could be implemented here? Is there any way to be enable the changes we need without conflicting the main branch? How should we start? I am just an user and willing to test, but @Thro42 might help with the actual coding, we just need some guideance :) This fw is too good to only be used with Marlin! 
Thank you for the help!

Thro42, 2021-01-09:
@bttguy I known what to do and I sen the first Pull Request. When its merged you can print files from Bord-SD-Card. Please let me known witch wiches you haf.

oldman4U, 2021-01-11:
Hi bttguy.

Sorry, but somehow I missed your answer.

I started to contribute a few months ago, learning all the apps needed to do so at the same time and still struggling to do everything the right way. But the procedure itself is very simple. You download the current version, edit it and push it to the current branch. For down- and uploading I use GitHub Desktop, but I believe also VSC can be used for this. If you have to make an extra branch or if you contribute directly to the master will depend on many things, including the amount of changes you want to make and how experienced you are.

Seems Thro42 knows what to do, so hopefully this will be a successful project!

Happy coding

bttguy, 2021-02-07:
@Thro42 
First I post my FR here for RRF instead a new one. So far appreciate the work that you made for RRF support :)
I would like to ask if it would be possible to implement the macro access for SBC users. [SBC setup example](https://teamgloomy.github.io/skr_pro_connected_sbc.html) This way a RPi4 runs the DWC and stores all the sys files and macros and connected to the mainboard via the EXP port. The current fw seems not to handle this setup. Thank you in advance!
Also a brief documentation would be handy about what features you implemented so far, and what you are plan/currently work on.

bttguy, 2021-03-11:
@bigtreetech Glad to see that you released a RRF board. With that, I would like to ask you if it would be possible that you also try to dedicate a fraction of your time helping in the work to make your awesome TFT-s more and more compatible with RRF?
As you can see since this FR started several developers already commited and moved their changes from the outdated forks to the main branch with all the upgrades and there are also some changes coming :) I can and will help with the readme/documentation side of the project. I think if you can also help manage this project we could achieve a better result, that would be beneficial for both us users, and for you as well! Thank you!

bttguy, 2021-03-11:
Also what do you think about making 2 config.ini to choose from? One for Marlin and one for RRF. It wouldn't need any fw mods in configuration.h, duplicate and restructure the config.ini which is only just an overwrite file for the fw. RRF compatibility requires some specific changes and lot of the Marlin settings are meaningless for us. If there is support from @bigtreetech and the main/older developers(@oldman4U you can help us in the thinking) I can try to make one from a RRF user perspective, but I am sure it would be very helpful for every RRF user. Just as the binaries are now autocompile and just needed to copied to the as card, a properly set up RRF config could help avoid errors, and let more people try the fw, with more success.

bttguy, 2021-03-12:
#1718

bttguy, 2021-03-22:
We can discuss RRF compatibility better in #1763 I guess, this will be a better place to collect the already implemented features, than actual talking.

closed 2021-04-03 by bttguy