bigtreetech/BIGTREETECH-TouchScreenFirmware #1403: TFT35 E3 V3 Meshbug

Doyle4, 2020-12-18:
Hello, I think I have found a bug and I am sure it is the screen firmware and not Marlin as its only started doing it with latest source files.

Before update, ABL menu, Mesh menu never showed up as bed levelling had yet to be done and saved.

Now when I power my printer, the mesh menu is empty and freezes the printer and a hard reboot is needed.

Any ideas?

Doyle4, 2020-12-18:
UPDATE: I have just done bed level on my printer and the Mesh Menu was still empty and froze my printer.

So after levelling the bug is still there.

oldman4U, 2020-12-18:
Hi.

Unfortunately I can not help with this, not using BL at the moment...

@digant73 is your man!

Doyle4, 2020-12-18:
No worries @oldman4U Thanks for tagging someone who maybe able to help 👎

digant73, 2020-12-18:
it seems you are using a tft version afffected by some bugs introduced by a PR. This bug was fixed in the last TFT version.
Try to use last FW (compile it. Don't use the precompiled version)

Doyle4, 2020-12-18:
Not compiled my own before so i'll have a try now.

Doyle4, 2020-12-18:
@digant73 That solved the issue thanks!
@oldman4U Thanks again for pointing me in the right direction!

closed 2020-12-18 by Doyle4

oldman4U, 2020-12-18:
The Pro helped you. I am just a poor boy....

😁

Happy printing