bigtreetech/BIGTREETECH-TouchScreenFirmware #1404: [FR] Add BUZZER parameters configuration block for Marlin (LCD12864) mode too

trengtor, 2020-12-18:
Hi! 

Could you add BUZZER parameters configuration block for Marlin (LCD12864) mode too?

It will very useful for custom hardware configurations with SKR mini E3/SKR E3 Turbo board seies.

oldman4U, 2020-12-20:
Hi.

2 days no answer.....

Do you mean the ability to define when a tone is played and when not? I believe this is not possible, because while using Marlin the TFT can not know which function calls the tone, but I add someone who knows Marlin very well.

@Tygrys-1 Are you still available to help with Marlin related questions or have you retired?

oldman4U, 2020-12-23:
@kisslorand 

Because you are working on the tone after a filament break... Can you see how this could work?

kisslorand, 2020-12-23:
Unfortunately my TFT doesn't work in Marlin mode so I cannot test anything in that mode.

oldman4U, 2020-12-24:
OK, thank you.

Well, I am still not sure if I understand the request right, trengtor... Do you mean the ability to define when a tone is played and when not?

Maybe radek8 can help;-(

closed 2021-03-02 by stale[bot]