bigtreetech/BIGTREETECH-TouchScreenFirmware #1408: [BUG] After stop not realse motors and not turn off bed heatrs

DanielSan1000, 2020-12-19:
### Description

Software ffter stop not realese motors and not turn uff bead heater

**Expected behavior**
Should relase everthing and turn off bed heater

### Hardware Variant
BIGTREETECH-SKR-mini-E3
BIGTREETECH-TFT35-V3.0

radek8, 2020-12-19:
I can confirm the error.
When you cancel printing, the hotend heating is switched off.
Remains on: Bed heating, cooling fan and stepper motors.

radek8, 2020-12-19:
You can solve this by enabling Cancel Gcode

cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\nM107\nM84\n

kisslorand, 2020-12-19:
This ticket should be closed, it's not a bug. Cancel gCode is there for a reason: to use and adapt it to your needs.

oldman4U, 2020-12-20:
Please help the community and close the ticket once you do not need it anymore. Thank you

DanielSan1000, 2020-12-20:
Thanks for help. Cancel g-code is working

closed 2020-12-20 by DanielSan1000