bigtreetech/BIGTREETECH-TouchScreenFirmware #1409: [BUG] During printing not possbile change paramters

DanielSan1000, 2020-12-19:
Description
During printing (when started) not possbile to change paramters
- change fan speed
- baby step
- print speed/load

Expected behavior
It should be possbile

Hardware Variant
BIGTREETECH-SKR-mini-E3
BIGTREETECH-TFT35-V3.0
Marlin 2.7.0.2

radek8, 2020-12-19:
Is the More icon not working for you?

DanielSan1000, 2020-12-19:
> Is the More icon not working for you?

Is strange because after some times it is possible to change parameters. Changes are not working immediately.

radek8, 2020-12-19:
you are allowed in Marlin?
REPORT_FAN_CHANGE (in Configuration_adv.h)

kisslorand, 2020-12-19:
It is possible to change them but it takes time. The most important thing during printing is the printing itself. Other messages are sent to the printer only  when it's possible. I agree, it can be improved, I also find it cumbersome to set the flow and speed during print. However Baysteps ar working just fine for me.

oldman4U, 2020-12-20:
The delay is a known issue. Also changing feedrate and so is delayed. Printing fron on board sd is even worse.

oldman4U, 2020-12-20:
Please see #1139 and please help the community and close the ticket once you do not need it anymore. Thank you

DanielSan1000, 2020-12-20:
Thanks for help

closed 2020-12-20 by DanielSan1000

kisslorand, 2020-12-20:
Even if it's closed, I started working on this. There's room for improvement, we'll see how much...

oldman4U, 2020-12-23:
DanielSan1000. Please give #1417 a try. It works GREAT;-)

DanielSan1000, 2020-12-24:
> DanielSan1000. Please give #1417 a try. It works GREAT;-)

I will check it and will give feedback