bigtreetech/BIGTREETECH-TouchScreenFirmware #1416: [BUG] Printing fails after running mesh bed leveling

jjdmon, 2020-12-21:
### Description

When I attempt to print after running mesh bed leveling the prime/purge strip will be under-extruded and the printer will run to the end of the y-axis and keep going with the servo drive gear skipping in the stopped belt.  It would probably stop at some point but I generally shut off the power before it does.   I use the encoder to adjust the height of the print head while doing the leveling and that also causes the print speed to be reduced to 10%, which is a known defect.  Disconnecting the EXP3 cable and running with only the serial cable connected eliminates the problem of the encoder changing the print speed but does not fix the problem of running off the y-axis.  It seems to happen pretty much every time I try to print after running MBL unless I power cycle in between.

### Steps to reproduce

1. Pre-heat printer
2. Check bed level at corners
3. Run mesh bed leveling and save mesh to EPROM
4. Start a print
5. Turn power off when the printer hits the end of the y-axis and tries to keep going

**Expected behavior**
Normal print after performing mesh bed leveling.

**Actual behavior**
Printer hits the end of the y-axis and keeps going while under-extruding the prime/purge strip.

### Hardware Variant
TFT board:  BTT TFT35 E3 V3
Main board: BTT SKR Mini E3 V2

### TFT Firmware Version & Main Board Firmware details
TFT firmware: Commit d8203d... from 12/17 in the BTT git repository for TFT firmware.
Marlin firmware: Commit 1fc9de... from 12/10 in the BTT git repository for SKR Mini E3 V2, Marlin 2.0.7.2

### Additional Information
Zip file of TFT and Marlin configuration files included.

[Config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5721927/Config.zip)

oldman4U, 2020-12-21:
Hi.

Could you please check if this solves your problem: https://github.com/digant73/BIGTREETECH-TouchScreenFirmware/tree/bugfix-2020-12-20

@digant73 is the master of this part of the code

jjdmon, 2020-12-21:
I tried the latest commit from that branch (1f6717...) and the problem is still there.  The symptom is similar to the y-steps being multiplied by a factor >1 after the leveling (except the speed doesn't seem to be affected), but not saved to EPROM since power cycling fixes it.  I actually checked the steps/mm in both TFT and Marlin mode after stopping the printer when the problem occurred and they looked ok.  Maybe it's temporarily changing the bounds of the print bed?

digant73, 2020-12-21:
it seems to me a bug in Marlin. I found a lot of bugs in Marlin FW 2.0.7.2. They are continuesly fixed with the bugfix release.

oldman4U, 2020-12-21:
Thank you digant73.

jjdmon, could you pease update to the latest Marlin version to see if this solves your problem.

digant73, 2020-12-21:
@jjdmon try this PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1415.
It should fix your problem

jjdmon, 2020-12-21:
I'm using the latest from the BTT git.  Not sure how far behind that is.

I did try the same procedure in Marlin directly: preheat, check corners, mesh level and print.  Worked fine with just Marlin in the terminal mode.  I tried again using the touch screen and it failed as usual.  After stopping the print I switched back to the terminal and tried printing again.  That failed the same way, so something is being set incorrectly on the Marlin side only when MBL is run from the touch screen.

jjdmon, 2020-12-21:
I pulled the "fixes #1416" commit (81c35a...) from the bug fix git and that does fix the issue.  Printing started normally after mesh leveling and did not run off the y-axis.

oldman4U, 2020-12-23:
Please help the community and close the ticket once you do not need it anymore. Thank you

jjdmon, 2020-12-23:
Shouldn't it stay open until the fix is merged into the master branch?  As of this morning, #1415 is still a pending PR.

oldman4U, 2020-12-23:
yes. why not;-)

PR #1415 (2020-12-24): Configurable XZ probing point on Z Offset menu (fixes #1363)
Based on further clarifications on bug #1363, The problem was not on Z axis but on XY axis.
On some HW configuration, the XY probing point set for homing is not reachable by the nozzle applying the XY offset configured on Marlin FW.
This PR allows to configure if nozzle must be moved or not on the XY probing point when the Z offset tuning process is started on Z Offset menu

fixes #1363 (definitively)
fixes #1416 (absolute coordinates and soft endstops were not restored after MBL process was finished)
fixes #1420
fixes #1425
closed 2020-12-24 by bigtreetech