bigtreetech/BIGTREETECH-TouchScreenFirmware #1419: Multi extruder Cold extrusion prevent

myunchulshin, 2020-12-22:
[image]

I am using a multi 4 color extruder.

Raise the hot end temperature and test Extruder No. 1, and it works.

After that, if you change it to 2,3,4, an error is displayed as shown in the figure.

I want to know how to fix it.

There were no errors in previous versions of the build.

guruathwal, 2020-12-23:
What type of multi extruder are you using? single hotend with multiple extruders or separate hotend for each extruder?

myunchulshin, 2020-12-23:
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5733591/Marlin.zip)
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5733595/config.zip)

Thank you 

음성듣기번역하기언어전환영어
Attached is the file.

guruathwal, 2020-12-24:
@myunchulshin  Can you check if #1427 solves your problem?

closed 2020-12-25 by bigtreetech

myunchulshin, 2020-12-29:
> @myunchulshin Can you check if #1427 solves your problem?
The problem was not solved.

PR #1427 (2020-12-25): Add coding standards, language pack size check during compiling, fix incorrect temperature for mixing hotend, reduce program memory usage
- Add Coding standards.
- Reduce code and program memory usage.
- Add Language pack size check during compiling.
- Fix incorrect temperature for mixing hotends / multi extruders with single hotend #1419 
- Update language pack size for TFT70
Fixes #1419