bigtreetech/BIGTREETECH-TouchScreenFirmware #1420: [BUG] ABL not saving mesh, current build as of 12-20-20

ki6wik, 2020-12-23:
Current build as of 12/20/2020 mesh not saving after running the ABL from menu. Invalid mesh error.
Pre compiled BIGTREE_TFT_28_V3.0.26.X.BIN
Also does not load mesh saved in marlin mode in unified menu mode. Was working from build 11/16/2020.

BTT SKR 1.3
BTT TFT28 v3

oldman4U, 2020-12-23:
Hi.

Have you already tried the build from digant73 from here: https://github.com/digant73/BIGTREETECH-TouchScreenFirmware/tree/bugfix-2020-12-20

Please let us know if this solves your problem

ki6wik, 2020-12-23:
Yes, this did correct the problem with the mesh not being saved and the error of invalid mesh.
Thank you 

From: oldman4U
Sent: Wednesday, December 23, 2020 2:51 AM
To: bigtreetech/BIGTREETECH-TouchScreenFirmware
Cc: ki6wik; Author
Subject: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] [BUG] ABL not saving mesh, current build as of 12-20-20 (#1420)

Hi.
Have you already tried the build from digant73 from here: https://github.com/digant73/BIGTREETECH-TouchScreenFirmware/tree/bugfix-2020-12-20
Please let us know if this solves your problem

oldman4U, 2020-12-23:
Purrfect👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you

ki6wik, 2020-12-23:
The bug fix took care of the issue of not saving the bed level mesh, resolving the invalid mesh error.

PR #1415 (2020-12-24): Configurable XZ probing point on Z Offset menu (fixes #1363)
Based on further clarifications on bug #1363, The problem was not on Z axis but on XY axis.
On some HW configuration, the XY probing point set for homing is not reachable by the nozzle applying the XY offset configured on Marlin FW.
This PR allows to configure if nozzle must be moved or not on the XY probing point when the Z offset tuning process is started on Z Offset menu

fixes #1363 (definitively)
fixes #1416 (absolute coordinates and soft endstops were not restored after MBL process was finished)
fixes #1420
fixes #1425
closed 2020-12-23 by ki6wik