bigtreetech/BIGTREETECH-TouchScreenFirmware #1423: [BUG] Unable to change filament during print

xhejtman, 2020-12-23:
I am unable to change filament during print. 
I pause print. I do unload filament, load filament. I'm back in the main menu -> cannot resume print right now. I request print, it asks for continue. I select continue. It does not print I have to select resume, but ok. I resume and Z axis is completely messed - it resumes the print in completely wrong layer. 

Using commit 9148ffe5708ab59e0b5590a4b0e67a44d6bc47d1.

oldman4U, 2020-12-23:
Hi.

9148ffe is only changing the ReadMe at the front page and does not change any code which is loaded to the TFT. Checking the two other tickets you opened I would recommend to reset the TFT using the reset.txt file OR the menu on the TFT, AND to upload the firmware recommended by digant73.

Check your 3 tickets after that and let us know.

Thank you

kisslorand, 2020-12-24:
I can confirm there's something borked about Pause/Resume during printing. In my case after Pause the printing head goes almost to Home then returns above the printing part. I have to press Resume and wait for a purge than hit Continue on a popup than hit Resume again. It's not really nice to purge above a printed part. 
This was observed while air-printing form onboard SD.

oldman4U, 2020-12-26:
xhejtman

Are you also printing from on board sd?

xhejtman, 2020-12-27:
nope, I print from SD from TFT

kisslorand, 2020-12-27:
I tested Pause/filament change/Resume from TFT USB, works perfectly for me. Used the latest TFT FW, runout sensor enabled and connected to mainboard.

kisslorand, 2020-12-27:
Tested with TFT USB, Pause/change filament/Resume working as expected. Paused, removed filament, after inserting new filament it loaded automatically and resumed printing automatically.

xhejtman, 2020-12-29:
So, pause/unpause returns to correct layer. However, as of commit 9148ffe5708ab59e0b5590a4b0e67a44d6bc47d1, menus are a bit rearranged, so I cannot run unload/load gcode - it is moved to the Move menu which is not available during Pause. I can extrude though and as it does not change layer height, it is OK, you only have to match pause retract length during changing filament.

oldman4U, 2020-12-29:
Like I wrote before. This commit only changes the documentation and not the
code itself.

xhejtman, 2020-12-29:
Sorry for confusion, I reference the commit as the HEAD I use. I do not mean that that commit is wrong in particular.

kisslorand, 2020-12-29:
So, can you change filament?

xhejtman, 2020-12-29:
technically, I can. But that nice menu for unload/load filament (via gcode m701/702) is not availabe for me during paused print, I can only extrude manually and it works.

kisslorand, 2020-12-29:
Yeah, I thought about Load/Unload to be available during print and Pause. What if Babysteps become Load/Unload button when print is paused? How does it sound?

xhejtman, 2020-12-30:
I think Babysteps are important. What about moving Extrude into Move menu and Load/Unload remove from the Move menu instead of the Extrude that is available during Pause? I makes more sense I guess. Extrude is just moving the extruder. Load/Unload it something different..

kisslorand, 2020-12-30:
I found an icon that could be replaced. When you press "More" there's a "Terminal" button. It can be replaced by "(Un)Load" since "Terminal" can be found also if you press "Machine".

kisslorand, 2020-12-31:
Check PR #1466

PR #1466 (2021-01-12): ExtrusionMenu-Load/UnLoadToMoreMenu
### Requirements

BTT TFT or MKS TFT

### Description

- added Load/Unload menu to "More" menu during print (replaces Terminal which can be also found under More -> Machine menu)
- "Extrusion" menu overhauled (preheat option if nozzle too cold, "Heat" button instead of "Nozzle" if only one nozzle, question about cooling down when leaving the menu)
- during print Pause, no question about cooling when exiting Load/Unload or Extrusion menu
- Load/Unload button moved to Mainmenu and Extrusion button moved to Move menu, following @mehmetsutas's logic
- rearranged [... cut]
closed 2021-01-12 by bigtreetech