bigtreetech/BIGTREETECH-TouchScreenFirmware #1424: [BUG] Wrong layer on display

xhejtman, 2020-12-23:
It seems, that after some updates (at least as of commit 9148ffe5708ab59e0b5590a4b0e67a44d6bc47d1), the current layer information is wrong and displays random values. E.g. on first layer it displays 0.6mm (instead of 0.2mm).
After pause/resume, it displayed 11.2mm instead of 1.2mm.

kisslorand, 2020-12-23:
"Layer" displays the Z position of the hotend,. You probably have 0.4 mm Z hop and 10mm Z raise at pause.
The TFT checks periodically the hotend's position, it is not a realtime info.

oldman4U, 2020-12-29:
Please help the community and close the ticket once you do not need it anymore. Thank you

xhejtman, 2020-12-29:
As of the commit 9148ffe5708ab59e0b5590a4b0e67a44d6bc47d1, Layer displays Z position including Z hop. It is still wrong, as the Z Hop is rare compared to Z printing possition, so I would expect to mostly display print Z position not Z hop. ESP 3D module monitors Z position correctly. Perhaps increasing Z monitor period helps?

kisslorand, 2020-12-29:
The problem is always present? You have wrong layer info the whole printing process?

xhejtman, 2020-12-29:
No, it is not always, but I would say 99% of time. For a bit, I can see the correct value, but mostly it shows the layer + z hop.

kisslorand, 2020-12-29:
I cannot think of any other issue than you have a low RX buffer in Marlin. However I might be utterly wrong...

xhejtman, 2020-12-29:
I can try. I suppose you are mentioning RX_BUFFER_SIZE. This is currently unset. What value should I use?

kisslorand, 2020-12-29:
It depends on what mainboard do you have. Minimum should be 4 if you are using a motherboard with 8 bit AVR processor and 16 recommended for 32 bit ARM.
I am talking about BLOCK_BUFFER_SIZE.

xhejtman, 2020-12-29:
This is already 32. I use SKR 1.4 Turbo board.

kisslorand, 2020-12-29:
Ah! I think I know the answer. The layer shown is the layer height of the last sent gCode (if you print from TFT USB or TFT SD). Since you have such a big buffer it takes time until Marlin will process the sent gCode. Lower you BLOCK_BUFFER_SIZE in Marlin to 16, 32 is overkill.

xhejtman, 2021-01-01:
Well, I decreaset the BLOCK_BUFFER_SIZE to 16, I would say, there is no change. However, during 3d benchy print, there is a lot of z hops and I believe the reason is as you say, it just displays the last height. So it is up to you whether you consider it a bug or feature.

kisslorand, 2021-01-01:
What if you print a regular cube?

xhejtman, 2021-01-01:
the layer is correct without z hops. just if there are too many z-hops, you see layer including z-hops almost all the time.

oldman4U, 2021-01-01:
Using the original Ender3 12864 screen, I saw many issues with layer hight and remaining time and so on, which have been depending on buffer settings and the gCode file itself. Because of this I always use a strange retract length and Z hop hight so I can easily identify if the shown value is inclusive or exclusive. Maybe this also helps you.

Having more communication could overload the bus/buffer causing problems like interrupted prints and quality issues. The current implementation works great, independent of the storage used for the print and you can eventually improve the data shown on screen, by adjusting the buffer values.

xhejtman, 2021-01-12:
Closing then.

closed 2021-01-12 by xhejtman

kisslorand, 2021-05-02:
The issue has been found.
Check PR #1894.

PR #1894 (2021-05-06): Layer height consistent display, MCU load reduction during print, tiny RAM and Flash usage reduction
### Requirements

BTT or MKS TFT

### Description

- consistent layer (nozzle height) display during print 
The TFT now shows correctly and in realtime the nozzle height. However, when printing from TFT SD or TFT USB, it shows the nozzle height based on the last sent gCode to Marlin, not the one that Marlin executes.

- code optimization (during print)
The progress percentage and elapsed time display procedure is slightly reworked, it's now a single procedure instead of two. The progress bar is [... cut]