bigtreetech/BIGTREETECH-TouchScreenFirmware #1426: [BUG] SKR Mini E3 1.2 + TFT35 V3.0 + Octoprint == zits on curved surfaces

pig-cop, 2020-12-23:
I know this was discussed before, but I just assembled my TFT35 V3.0 and noticed this problem on the first print.
Apart from disconnecting the TFT (which is a workaround) is there any progress on this?

I saw all the previous posts are closed, so I wonder if I missed some update that resolved this, or it's still an issue?

oldman4U, 2020-12-23:
Hi.

Old posts are closed in case there has been no activity for a certain time. Verified Bugs and feature requests (also from closed tickets) are collected in the corresponding pinned tickets.

I can not say how many users here are using Octoprint and if someone has a solution. But assume you are using the latest version of Marlin and the TFT firmware from this week.

Maybe you can upload a movie showing the issue.

pig-cop, 2020-12-24:
Never mind, updating Marlin to 2.0.7.2 and the TFT35 to this week's firmware resolved completely the issue.
It was silly of me not updating all the firmwares before posting!
Thanks!

closed 2020-12-24 by pig-cop

oldman4U, 2020-12-24:
Thank you for letting us know 

Happy printing and Merry Christmas

pig-cop, 2020-12-24:
Merry Christmas to you too and all the fellow makers here!! 🎉