bigtreetech/BIGTREETECH-TouchScreenFirmware #1429: [Q] MBL (manual) Zoffset code does not change from M851 to G29 S4

karabas2011, 2020-12-24:
Marlin 2.0 from 23.12.2020
BTT TFT from 15.12.2020
No Zprobe on my printer.  Manual MBL 3x3
Q:
Babystepping still uses M851. Why?? It need to be G29 S4 Z
Mesh editor when editing point moves beyond printing area. and I prefer manual correcting.

As I see the solving may go at least 2 ways.
1 Change all M851 Z  to G29 S4 Z and imitate Zprobe.
Also change marlin response G29 S4 (make it as M851) for easy parsing 

2 Write a new code with "MBL Adjust".

oldman4U, 2020-12-24:
Please try the TFT firmware from today which likely included a fix.

karabas2011, 2020-12-24:
Ok. I have dl and tried. Nothing changes.

MBL shows OFF while it is actually on.
Zoffset still tries M851 
Mesh editor goes beyond Z area

digant73, 2020-12-24:
With mbl enabled in marlin, z offset is not automatically changed when you change babystep. For that reason it is displayed in red in babystep menu. If you save from the menu the z offset is saved.
The mesh editor simply moves the nozzle to the set z value. It seems to me you probably changed z offset after you performed mbl process

karabas2011, 2020-12-25:
I cannot do anything without hacking because
1 it ignores the absence of Zprobe and tried to define probing  position or something.
2 After G28 it tried to raise nozzle to 10mm and crash the bed.
3 Then it moves nozzle to probe position and crash it into bed.

so I hacked the code removing all this stuff and move to M206 in Zoffset menu

oldman4U, 2021-01-01:
@digant73 Anything you have to change in your code?

digant73, 2021-01-02:
@karabas2011 can you provide all the files you changed just to understand better your setup?

sarvenn, 2021-01-10:
Is prebuild bin file updated for this issue?
I have skr mini e3 v2 & tft e3 v3 on my ender 3 pro.
I've installed prebuild firmware on my motherboard and I use manuel mesh bed leveling. I saw the same bug at z offset menu. M851 unknown command error. Can not adjust Z offset. I only can adjust Z offset while printing with babystep Z. I also save it to eeprom while printing however it doesn't updates the Z offset while I send M503.

karabas2011, 2021-01-11:
I will dl latest snapshot, modify it and post result code

karabas2011, 2021-01-11:
[fixM206.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5794684/fixM206.zip)
My previous fix for Zoffset menu.

digant73, 2021-01-11:
I had a look to your changes but the problem is that you need Home Offset instead of Probe Offset. Your problem is not related to MBL. A Home Offset menu is not currently provided by the TFT fw.

sarvenn, 2021-01-11:
@digant73 hi,
I use prebuild firmware.bin for my skr mini e3 v2 (latest prebuild firmware is installed from btt repository). I use manuel mesh bed leveling. I also have TFT35 e3 with the latest firmware installed.
I can successfully do the mesh bed leveling and save the values to epprom. I checked from M503 command.
However I only can set the z offset at marlin mode. Z offset in touch screen mode does not work.
So while printing I use babystep feature at my first layer and adjust the babystep, save to epprom but it never updates the G29 z offset value. Also after print is finish I go to menu>movement and babystep, my adjusted value can not be seen there. It is always 0. I recorded a video. I'll upload it soon.

I think touchscreen firmware should have variants for bed leveling option just like the prebuild motherboard firmwares (manuel bed leveling, bl touch, bl touch+z homing). I think my tft firmware tries to send bl touch specific commands and also doesn't save my babystep values.

In marlin mode I adjusted my bed z to 0.05 but I always print from TFT mode. I don't know if my babysteps add on that bed z value. I expected that my babystep values update my bed z value. And also Z offset function should work as well.

sarvenn, 2021-01-11:
While printing adjusted babystep
[image]

sarvenn, 2021-01-11:
https://user-images.githubusercontent.com/66178082/104218854-6397b100-544e-11eb-90b6-1c838a53c333.mp4

sarvenn, 2021-01-11:
At marlin mode my bed z value is -0.05.
You may see my babystep value is reset to zero after printing is finish.
Z offset unknown command. etc.

https://user-images.githubusercontent.com/66178082/104220424-83c86f80-5450-11eb-8e86-64db452212e2.mp4

karabas2011, 2021-01-12:
> I had a look to your changes but the problem is that you need Home Offset instead of Probe Offset. Your problem is not related to MBL. A Home Offset menu is not currently provided by the TFT fw.

Yes, no home offset menu for probeless setups. So I made Probe offset works insteed. It must me recoded in future to common approach. And I replaced initial M581 with M503. Now offset is read when you press  Z offset button.

digant73, 2021-01-12:
@sarvenn the problem you have is that you configured Marlin to use Home Offset instead of Probe Offset.
In Babystep menu you save the home offset.
It is correct that babystep is reset to 0 every time G28 is executed. Having babysteo 0.05 with z offset 0 is equivalent to babystep 0 with z offset 0.05. But in your case babystep menu saved Home Z offset instead of Probe Z offset

karabas2011, 2021-01-12:
I  didnot save babystep currently. Only home offset

sarvenn, 2021-01-12:
> @sarvenn the problem you have is that you configured Marlin to use Home Offset instead of Probe Offset.
> In Babystep menu you save the home offset.
> It is correct that babystep is reset to 0 every [...]

Hi @digant73,
I didn't configure Marline myself. I used the prebuild firmware.bin file from BTT SKR mini e3 v2 repository. I don't use a bed leveling prob, so I installed the relevant prebuild firmware, not the bl touch ones. I also use prebuild TFT35 E3 firmware. My firmwares are up to date.
Does this mean that prebuild firmware need to be re-configured accordingly to solve the issue? If so, then this issue should be cascaded to motherboard developers, right?

How can I solve my issue in TFT mode?

digant73, 2021-01-12:
@sarvenn A home offset menu should be provided instead of probe offset menu (as made by karabas).
@karabas2011 I understood. Basically you needed a home offset menu (currently not provided by the TFT fw). If I will have some time I would make that. Not a promise (I'm full of work in this moment)

sarvenn, 2021-01-14:
@digant73 @oldman4U Could you include this item in bug fix list if appropriate.

digant73, 2021-01-14:
I'm implementing the Home Offset menu based on the info provided by @karabas2011. It will requires some testing from you (I don't have home offset set in Marlin fw).
Do not close this thread

karabas2011, 2021-01-14:
About manual MBL - similar problem:  head tried to move 10mm beyond endstop and when going to probe crash into bed.
So I hacked it similar way to comment out corresponding commands .

digant73, 2021-01-14:
Which source code did you change for that? ProbeHeightControl? Please attach the changed files. Also you reported in the thread title to use G29 S4 instead of M851 but in your last code you used M206 (home offset). Was it a mistake?

oldman4U, 2021-01-14:
A week ago we added some information to the ReadMe about connecting the MKS to a TFT.

For me it seems, that ABL, home/probe and Z-Offset, Babystep, MeshEditing and so on, is a much bigger issue for many users. Wouldn't it be good to have some documentation about this also?

digant73, 2021-01-15:
@oldman4U The problem is only the need to use Probe or Home gcodes in MBL. I just added the Home Offset menu according to the code provided by @karabas2011. I need just the changes @karabas2011 made also for MBL menu so I will integrate them. Unfortunately I don't have Home enabled in my Marlin FW to test it by myself

sarvenn, 2021-01-15:
MBL= Manual mesh bed leveling? Without a probe or bl touch, just setting the nozzle height.
I ask it because I want to be in the same page with you while reading your comments @digant73 . Just wanted to confirm my knowledge.
As I figured out, in marlin mode when I change the bed Z to adjust my nozzle height to achieve a perfect first layer, and save that value to epprom, it changes M29 S4 parameter. Since I use the precompiled firmware for my SKR Mini E3 V2 in BTT repository, when clicking on the Z offset button on my TFT35 it sends M851 which is not recognized by the motherboard. I have uploaded a video above.
To avoid firmware complexity for TFT (manual mesh bed leveling, bl touch etc.) firmware I think that it would be a good idea to add another menu to adjust the Z offset for manual bed leveling users. May be a selection option could be added to feature page asking for if I am using Probe or Manual for bed leveling to show me relevent Z offset menu. Is it possible?

oldman4U, 2021-01-15:
digant73 on steroids;-)

Please check #1518 

Thank you

digant73, 2021-01-15:
@sarvenn, @karabas2011 
Please try PR #1518. Now the Z Offset menu will rcognize if probe or home offset is supported by Marlin FW.
In your case, you should now see a Home Offset menu while in the past it was always Probe Offset.
Also please test MBL menu and Mesh Tuner Menu and let me know if you your feedback (I suspect that you will complain of that the nozzle raises too much) so I will apply proper changes also on that menus

@sarvenn MBL menu is applicable to both probe/home features not just home as I suspect you think about it. The same for the Mesh Editor

sarvenn, 2021-01-15:
@digant73 so fast response. Thanks.
Does this improvement included in the prebuild firmware? I don't compile myself, only use the prebuild ones.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
Otherwise how can I install and try your improvement to give you feedbacks?

Oh you pointed a good one. Yeah I noticed that while doing a manual mesh bed levening sequence, On some points nozzle touches presses to the bed which may cause the springs to compress (so that manual bed leveling with knob may ruin), after that nozzle raises and wait for my adjustment. You mean this I guess, right? I would be happy to try this as well at your new firmware.

digant73, 2021-01-15:
the sequence followed by MBL and Mesh Tuner is:
1) Homing (G28)
2) raise the nozzle
3) move the nozzle to the desired probing point (for MBL the nozzle is moved to Z 0.2 mm just to avoid to touch the bed)
4) you can adjust Z offset
5) repeat steps from 2 to 4 for the other probing points (for MBL)

Please let me know which of the above steps give some problems.

You must download my PR and compile it for you TFT version

digant73, 2021-01-15:
Also, in babystep menu, when you change babystep do you see also z offset changing?

sarvenn, 2021-01-16:
Hi @digant73 ,
I visited your PR. In this directory there is precompiled firwares. May I use them instead of compiling myself? Is it possible?
https://github.com/digant73/BIGTREETECH-TouchScreenFirmware/tree/bugfix-2021-01-15/Copy%20to%20SD%20Card%20root%20directory%20to%20update
These firmwares seem to be updated recently. Forgive me if I bother, but I have never compiled a firmware on my own before. I know it is not a big deal. However I rather prefer to use a compiled firmware from a developer's hands :)

Codel1417, 2021-01-16:
@sarvenn the firmware only compiles in the copy to... folder when something is pushed to master of the bigtreetech repo. Compiling yourself is easy. all you need is 'visual studio code" and the 'platformio' extension

sarvenn, 2021-01-16:
@digant73 I comment on your PR.

karabas2011, 2021-01-17:
Thanx I will try it

oldman4U, 2021-01-19:
karabas2011

Have you been able to test digant73's PR in the meantime?

PR #1518 (2021-01-21): Added Home Offset menu + bugfixes and cleanup
**NEW FEATURES:**
1) Added Home Offset menu. Previous ProbeOffset menu is now a generic Z Offset menu (just to not duplicate code) that will invoke Home Offset or Probe Offset API. Both the Home and Probe Offset menus are available on the Bed Leveling menu. Furthermore, Probe Offset menu is available only if probe is enabled (supported) in Marlin fw.
2) Added support to MBL Z offset (G29 S4 Zxx) in Babystep menu. Babystep menu will allow to change/save Z offset based on the following rules (matching sequence):
 *) Probe Offset: If probe is supporte [... cut]
closed 2021-01-21 by bigtreetech