bigtreetech/BIGTREETECH-TouchScreenFirmware #1430: [BUG] (short description)

AntoszHUN, 2020-12-24:
Does resume printing after "Pause" work for anyone else?

[image]
[image]
[image]

I pause printing. Then I wanted to continue after that. The machine pushed the head into the print. Something is not right. In previous firmware, it still worked.

oldman4U, 2020-12-24:
Hi.

Can you try to find which PR broke this?

AntoszHUN, 2020-12-24:
I searched. But I don't know which statement could have changed it.

oldman4U, 2020-12-29:
There have been some issues with config.ini. Try it again and let me know. Are you printing from on board sd? Maybe a movie would help.

AntoszHUN, 2020-12-29:
> There have been some issues with config.ini. Try it again and let me know. Are you printing from on board sd? Maybe a movie would help.

All right. I'll see as soon as I have time. Yes, I print from a TFT card reader.

oldman4U, 2021-01-01:
Any news related to this issue my friend?

AntoszHUN, 2021-01-02:
> Any news related to this issue my friend?

I updated the firmware. but I haven't tried the pause yet.

oldman4U, 2021-01-02:
Using the current firmware for the TFT

https://user-images.githubusercontent.com/54359396/103466309-322f2f00-4d44-11eb-8028-39ebc4f08ade.mov

AntoszHUN, 2021-01-02:
> Using the current firmware for the TFT
> 
>  Pause.Print.mov

It seems good to me too. I will also close the topic. Thank you!

closed 2021-01-02 by AntoszHUN

oldman4U, 2021-01-02:
Works?

oldman4U, 2021-01-02:
Perfect!

👍🏻