bigtreetech/BIGTREETECH-TouchScreenFirmware #1432: [Q]  How to correct error Error: unable to find CMSIS-DAP device

doug-scott, 2020-12-24:
I have a BTT TFT35-E3 V3.0 device.  I have downloaded the firmware from root page of this repository by selecting the CODE/Download zip option.  I extracted the zip file, and then launched Visual Studio Code app and opened the folder from the File menu.  Once the newly created folder was opened I edited the platformio.ini file to reflect which ENV I was going to build.  The default_envs now says this:

default_envs = BIGTREE_TFT35_E3_V3_0

I then ran the PlatformIO Upload process from the icons under the Folders pane.  It compiles the firmware with some non-fatal warnings, but ends with this error message in the terminal window:

Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
*** [upload] Error -1

This version of Visual Studio Code is the same install that I use to compile Marlin 2.0 firmware(different workspace, not run at same time) and I do not have any issues with building with the upload tool.

How can I correct this error?  I currently have the SD card inserted into my PC, and do realize I could just locate the newly build bin file over myself, but I would much rather no having to do that.  I sometimes forget to eject the SD card correctly and if done at the wrong time the SD card becomes corrupt, and requires to be formatted.  This SD card is a 64gb card and having to format it into a small partition in FAT32 can be troublesome sometimes.

guruathwal, 2020-12-24:
Thats because you are clicking on upload button in platformIO. You need to click on build.

doug-scott, 2020-12-24:
Upload builds the firmware, then uploads it to the device that is specified somewhere.  When building Marlin I also click on Upload, and it builds the firmware then uploads it to the SD card that is configured with a volume name of REARM, and formatted in FAT32 with a size less than 32gb.  I am not a programmer, but this must be defined within the Marlin source code (the upload function I mean) somewhere.  I know the Upload_Protocol is not specified in the platformio.ini file for the Marlin boards, they don't even use that declaration.  I tried the same idea in platformio.ini for this firmware but it gives this error

Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

I think that is simply saying it needs something defined.  I have also tried changing it to use STLINK and it gives the exact same error about CMSIS-DAP device.

If this is something that is defined in the source code for the firmware like it appears Marlin is, couldn't that routine not be just implemented into this source code?

oldman4U, 2020-12-24:
Hi doug-scott.

Independent how it works with other soft- or hardware, updating the firmware of the TFT you have to follow the procedure described here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#update-tft-firmware

No ST-Link or something else is needed but a 8GB (or less) FAT32 formatted sd card and the firmware on it.

Happy printing

doug-scott, 2020-12-24:
Thanks for your input.  As I said I am not a programmer.  I did however spend many years working in ITG/QC and QA for a fax and modem sharing software company.  One of the parts I was responsible for was how the customer experience was.  This sort of thing for example.  The end user was the best resource for picking the GUI apart.  Developers would not test what they just checked in by using the GUI.  They would launch the their code from within the IDE since it did not require them to rebuild the client in order to check it.  
Imagine the results of having the build put the one correct bin file you need, along with the config.ini file, and the proper fonts and icons put into the correctly named folder.  They could also require the SD card volume to have a specific name.

Is this firmware part of the BIGTREETECH hardware manufacturer, or is it maintained, updated and managed by contributors who are not paid for their time or work?

oldman4U, 2020-12-25:
Hi. I am not a programmer too, but product manager in the broadcast industry. My son started with 3D printing 1.5 years ago and I am his technician;-)

The hardware comes from BTT, everything else from people like you and me and of course the hero's which are writing the code.
;-)

doug-scott, 2020-12-25:
I retired a number of years ago and am now too old to start learning the various programming languages.  I have a hard time with short term memory, even typing questions or answers will take me a couple of passes through the message because something distracted me for a couple of seconds.  You know your memory is bad when you read a post, you completely understand what the poster was trying to say, only to find out you wrote that post the day before.  
My question has been answered, I will use that button below now.

closed 2020-12-25 by doug-scott