bigtreetech/BIGTREETECH-TouchScreenFirmware #1433: [BUG] Screen freezes in Marlin mode while starting to draw status screen.

kad, 2020-12-24:
### Description

I have two TFT35 E3 V3.0 and both after recent firmware upgrade started to behave problematically.
Reverting to 1bd14113e088aac2430f2b8152784e361bebdec0 state helps.

Overall: display is used in touch mode - it is ok. But if it is in Marlin mode and printer boots with both EXP3 and 5pin TFT cables connected - once boot sequence is over and marlin starts drawing status screen (about 1/4th) the TFT freezes. it is impossible to change something or click something, impossible to change modes.

### Steps to reproduce

1. latest master build
2. SKR mini E3 board, 2.0.7.2 release, CR10_STOCKDISPLAY config
3. connect by both EXP3 and TFT cable.
4. Boot with Touch mode by default. verify that it works (moves, printer info, fan control, etc).
5. Switch to Marlin mode
6. Press reset button to reboot.
7. Observe boot sequence in marlin mode (custom boot screen, marlin boot screen, status screen).

**Expected behavior**
No freezes, correct drawing in Marlin mode.

**Actual behavior**
If booted with Marlin mode:
Custom boot screen shows, Marlin boot screen shows. Status screen starts to appear and display freezes.
Screen shows 1/4 of status screen, rest is from Marlin boot screen. No response on clicks.

If booted with Touch mode: it works 90% of times. Some times, after boot trying to initiate move commands from menu (e.g. Home), leads to screen freeze.

I tried to revert firmware to commit 1bd14113e088aac2430f2b8152784e361bebdec0 It works without issues.

### Hardware Variant
TFT3.5 E3 V3.0

### TFT Firmware Version & Main Board Firmware details
Marlin 2.0.7.2, bugfix-2.0.x same effect.

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

kad, 2020-12-24:
[image]
[image]

radek8, 2020-12-24:
Try disabling it in the config.ini file by selecting
serial_always_on: 0

kad, 2020-12-24:
> Try disabling it in the config.ini file by selecting
> serial_always_on: 0

as temporary workarround I'll try, but that's not a real bug fix.

oldman4U, 2020-12-24:
There are two dots in the upper corners. Left shows that there are messages available at the touchscreen (which messages) and the right shows that there is activity going on.

What else is connected and can you identify a specific TFT firmware version where this is broken.

I have no SKR mini but a SKR E3 DIP and have been using Marlin mode today with the TFT firmware from today.

radek8, 2020-12-24:
I agree that this is not a permanent solution.
More and more users are starting to respond with this error.
I didn't notice this error with me before, but today I found out that I have similar behavior on my printer.
If the option: serial_always_on is activated
and the TFT turns on in simulated merlin mode, so the TFT freezes.
I will try to intercept the communication to find out which communication the TFT freezes.
After switching to the classic display, Marlin is functional. So it is not Marlin that will freeze, but TFT

oldman4U, 2020-12-24:
Hi radek8

So serial always on has to be activated and you have to boot up in Marlin mode to verify this issue, right? If one of the two "parameters" is not true, the problem will not occur?

radek8, 2020-12-24:
The problem occurs if the option is activated in the TFT
  serial_always_on: 1

kad, 2020-12-24:
> What else is connected and can you identify a specific TFT firmware version where this is broken.

The one which I today built/flashed is 6d87753f436f71ec54951b41ae3391878fc2b60c
The one which I reverted to and tried is 1bd14113e088aac2430f2b8152784e361bebdec0

I do have in my config "serial_always_on" enabled, as that was recommendation for using those TFTs with printers connected to OctoPrint.

oldman4U, 2020-12-25:
Hi kad.

So there are 14 days between the two versions and PR merged on Dec. 15th, 17th, 19th and 23rd. Would be good to know which PR exactly broke it, so the developer can review his code.

serial always on is needed to keep the serial port active while using Marlin mode.

radek8, 2020-12-25:
The same problem
#1118
#1113

oldman4U, 2020-12-25:
But those two are from September and had writes that 1bd1411 works, which is 2 weeks old.

Btw, I believe you are in the same timezone. Maybe we should try to get some sleep;-)

radek8, 2020-12-25:
I tried 1bd1411
after 10 restarts it froze 5 times and worked 5 times

oldman4U, 2020-12-25:
Looks like a kind of race condition to me. Guess we need the pros for this.

guruathwal, 2020-12-25:
did anyone test with any other printer/board that uses both EXP1 & EXP2 ports with same marlin version and TFT firmware version?

oldman4U, 2020-12-26:
Tested with only EXP3 cable on SKR E3 DIP and TFT35 E3 and no problem. Will test with SKR v1.4 next.

oldman4U, 2020-12-26:
Can someone confirm the following which I can see on 2 different printers now.

I compile the firmware using VSC and PIO as usual and only change the env in platform.ini to match the TFT. Then I copy the firmware together with an unchanged config.ini and the TFT folder, both from Copy to SD Card root directory to update folder and update the firmware. Even the default baud rate is defined in config.ini, I get a No printer attached and checking the baud rate I can see, that there is NO baud rate selected in the corresponding TFT menu. Selecting the baud rate there works. 106 keyword found during update. Seems something broken here.

radek8, 2020-12-26:
Oldman, I can confirm the same behavior.
After updating to the latest FW and saving config.ini, the printer is not connected.
No baud rate selected on TFT. after manual selection, the printer connected.
Config.ini, ie does not save the baud rate, but deletes it.

kad, 2020-12-26:
> I get a No printer attached and checking the baud rate I can see, that there is NO baud rate selected in the corresponding TFT menu. Selecting the baud rate there works. 106 keyword found during update. Seems something broken here.

I think this is diverging from original issue and deserve a separate bug report about baud rates.

guruathwal, 2020-12-26:
This may be related. If the settings are being read incorrectly it may also be the reason for the freeze. it needs a little inspection to rule it out.

Roughd1433, 2020-12-26:
Hi
I just updated my tft34-E3_V3 with the new firmware and had the same issue
with  no printer attached msg. WHen I checked in settings there was no baud
rate selected. Manually selecting the correct rate then showed the printer
attached

guruathwal, 2020-12-26:
Fixed the no baud rate issue in #1438. Can anyone check the freeze issue after using #1438?

radek8, 2020-12-26:
I tested PR # 1438
- Temperature update is fixed.
- The problem with the communication speed beeing unset is still present after the update.
- TFT freezes after loss in simulated marlin mode and activated serial_always_on. It does not always freeze, out of 10 restarts it freezes about 2x.

guruathwal, 2020-12-26:
The baud rate issue should be fixed now with  https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1438/commits/e10488639136de98882641617dbd4a96c583d9bf

oldman4U, 2020-12-26:
Can not reproduce it. Also not using SKR v1.4 and Pro with TFT24/35/43.

Seems my v1.4 is broken now🥴

kisslorand, 2020-12-26:
> I tested PR # 1438

> * The problem with the communication speed beeing unset is still present after the update.

Make a reset.txt file on your SD Card.

I tested this, I had no initial baudrate problem until I used the reset.txt file. I had random issues when config.ini was changed (add/remove parameters) and reset.txt always cured it for me.

radek8, 2020-12-26:
After a new addition
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1438/commits/e10488639136de98882641617dbd4a96c583d9bf
I have the same problem as Oldman4U
The speed is saved correctly, but the printer is still not connected.
Resetting to factory settings will not help.

oldman4U, 2020-12-26:
In this case only I have the problem hopefully. During my tests the screen of the TFT24 became darker and darker and now there is nothing shown n screen anymore - when using the SKR v1.4. Using the same screen on a SKR Pro of E3 DIP works fine;-(( I bought the v1.4 only for testing and now after 3 weeks, it is broken...

oldman4U, 2020-12-26:
kad. Does disabeling "serial_always_on" solve the problem for you? Any which cables are you using, original or self made?

guruathwal, 2020-12-27:
@oldman4U with SKR 1.4 did you enabled REPRAP_DISCOUNT_SMART_CONTROLLER?

oldman4U, 2020-12-27:
Yes. Everything was working fine. I could make the Marlin test without a problem and suddenly the TFT became darker and darker. Will check the power output today.

kad, 2020-12-27:
> kad. Does disabeling "serial_always_on" solve the problem for you? 

can't confirm right now, as for temporary workarround I reinstalled stock display. 12864 is the only mode that I need in 99% of time due to OctoPrint, thus it is more important to have printer running stable than fighting with tft firmware.

> Any which cables are you using, original or self made?
Original 10pin IDC cable and original 5 pin dupont/jst 2.54 cable for rs232.

oldman4U, 2020-12-28:
In this case it is very unlikely that the problem will be solved. You opened this ticket, so it would be good if you could take care of it, or radek8 wants dot do it for you.

I can not reproduce the issue using TFT24/35 or 43 and SKR v1.4 as well as SKR Pro. Sounds more like a cable/hardware issue to me, but it wouldn't be the first time that I can not reproduce an issue using my test systems.

cirorebelo, 2020-12-30:
I had  a similar-ish issue I reported [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1352).
Maybe related maybe not. On SKR mini E3 v2, TFT would freeze if I booted on touch mode, went into Marlin mode and then back to touch.
Setting Serial to 'always on' solved that, for the time being. 

After a few more updates to the pre-compiled, using 'serial always on' from then on, I noticed the same issue reported here. I Had to boot while in Marlin mode and the screen froze during boot, at the same point, with the same display artifacts.

I had to reflash the TFT to get it out of the loop and boot into touch again. (Later I found out i could change modes if I was fast and did it during boot screens, before the freeze)

Then I tried to set 'serial always' to off and it solved the issue and, with the newer firmware, the old issue on my link did not come back. So I had no issues, and 'serial always' was set to off as it had been before the 1st issue appeared.

I may try to set always on back again to help diagnose, but I do not need it (was using as 'fix') nor do I use any other devices on the TFT's serials. (no octo, no pronterface).

Another thing I noticed that may be related to the 'settings not applying' you are looking into: 

On the same firmware The 'knob color' I set on config.ini was not applying using the previous firmware (before the one available now - it was there for 14days or so.) I had to set it on the TFT. 

Setting 'knob idle 0' solved that, don't know why. It wasn't going zero brightness, it had no color set, despite the config.ini setting one.

effgarces, 2020-12-31:
I'm having a similar problem with my TFT24. If serial_always_on is enabled then when entering Marlin mode the screen freezes and in my case I just get a blank screen, yet using the encoder seems to be working, as I can hear it clicking. I'm using it with a SKR 1.3 with Marlin 2.0.7.2, with every recommended option turned on and an ESP01 running ESP3D.
This exact same setup was working on a previous version of the firmware.

radek8, 2020-12-31:
> I'm having a similar problem with my TFT24. If serial_always_on is enabled then when entering Marlin mode the screen freezes and in my case I just get a blank screen, yet using the encoder seems to [...]

Yes, only the TFT display freezes.
The board with Malin continues to work and responds to the encoder.

kad, 2021-01-02:
@oldman4U tried to flash recent build from master (5b66d674bcb8e7faefdeda90e7ce812bf3464850) situation became even worse.  With serial_always_on=0 it works, but expectedly not able to communicate with printer from e.g. esp-01 wifi module in LCD12864 mode. with serial_always_on=1 TFT boots in touch mode. but if switched to lcd12864 and trying to reboot - hangs. Only title "LCD12864 Simulator" is visible. not even Marlin boot logos appears. Tried to reboot about 10 times, always consistent hang. Board where it connected to is SKR mini E3 V2.0, today's Marlin bugfix-2.0.x branch build.

radek8, 2021-01-02:
> @oldman4U tried to flash recent build from master ([5b66d67](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/5b66d674bcb8e7faefdeda90e7ce812bf3464850)) situation became even worse. With serial_always_on=0 it works, but expectedly not able to communicate with printer from e.g. esp-01 wifi module in LCD12864 mode. with serial_always_on=1 TFT boots in touch mode. but if switched to lcd12864 and trying to reboot - hangs. Only title "LCD12864 Simulator" is visible. not even Marlin boot logos appears. Tried to reboot about 10 times, always consistent hang. Board where it connected to is SKR mini E3 V2.0, today's Marlin bugfix-2.0.x branch build.

I can confirm this behavior.
If serial_always_on is enabled: 1
I switch to simulated Marlin mode, restart the printer, and the screen freezes after the restart. You can't even switch back to touch mode.
The only way to get out is to reload config.ini with the option
default_mode: 1
or
serial_always_on: 1

oldman4U, 2021-01-02:
Good news. I can reproduce the issue here. Maybe @guruathwal can help us.

kad. "serial always on" defines if the serial ports are active or not while using Marlin. So turning this function off, will also disable the ESP.

kad, 2021-01-03:
> kad. "serial always on" defines if the serial ports are active or not while using Marlin. So turning this function off, will also disable the ESP.

I know, but that's was exactly my use case: using LCD12864 most of the time, but keep serial open, so I can use ESP module if needed. That's why workaround with "serial_always_on=0" is not good for me (as commented above).

oldman4U, 2021-01-03:
All I can do, is to add it to the list of verified issues at the moment.

Hope someone will help soon.

guruathwal, 2021-01-03:
it may be the long M503 response during boot, can anyone check if the issue is solved with #1468?

effgarces, 2021-01-03:
Unfortunately the issue is still there, compiled with #1468 applied, reset the tft with reset.txt, applied new config.ini, reboot a couple of times with diferent settings, as well as hardcoded firmware settings, always the same result.

radek8, 2021-01-03:
The problem exists even if no printer is connected.
Just connect only + 5V and GND, restart in simulated mode and the TFT will freeze.

joelguth, 2021-01-03:
Can confirm the issue and same experience. The only way for me to get the screen to operate again is by reloading a new config.ini as @radek8 mentioned above.

kisslorand, 2021-01-03:
Does anyone know a build that worked?

radek8, 2021-01-03:
I noticed that the option in config.ini does not work on my display
marlin_fullscreen: 1

Can't it be related to the problem?

kisslorand, 2021-01-03:
Dunno. A working build might shed some light on it.

guruathwal, 2021-01-04:
Can you check if the issue exist with this  commit https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/8632baad030f497f78a173ba73f2ee7a25f5a760 ?

radek8, 2021-01-04:
Commit https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/8632baad030f497f78a173ba73f2ee7a25f5a760
appears to be functional without the problem of freezing in simulated Marlin mode

guruathwal, 2021-01-04:
Found the cause. The serial connection was not getting initialized while booting in marlin mode after PR #1438. 
After fixing this issue and booting in Marlin mode with`serial_always_on` enabled, I encountered two old issues. #1383 ("Click to Resume" message in marlin mode) and #1297 ("damaged M105 command"). Both are the same issues and are caused by `M105` being bombarded on Marlin firmware by the TFT when booted in Marlin mode with`serial_always_on` enabled. The `M105` was being sent continuously because the delay to resend the command was set to **0** in PR #1244. As a result, the Marlin firmware interpreted `M105` as `M1` because its buffer was being overwritten faster than Marlin could interpret it. Resetting the delay to its original value resolved these issues.

radek8, 2021-01-04:
Thank you
Your patch #1468 works fine

radek8, 2021-01-05:
guruathwal I will just specify that the problems of "Click to Resume" and "damaged M105 command" occurred in Marlin mode, but also after starting in touch mode

PR #1468 (2021-01-07): save/load infoSettings Struct instead of save/load every variable individually, Fix Screen freezes/Click to Resume message/damaged M105 command in Marlin mode
- The settings are stored in the last 2KB of the program memory. earlier a separate entry was used to save/ load each variable. With this the **infoSettings** Struct is copied at once to/from buffer instead of saving/loading every variable individually, removing the need to add save/load entries for each variable. This also eliminates bug related to wrong/invalid settings being loaded. The prebuild settings size check now works accuratel [... cut]
closed 2021-01-07 by bigtreetech