bigtreetech/BIGTREETECH-TouchScreenFirmware #1434: [BUG] Bad hotend and bed temperature update on status screen.

radek8, 2020-12-25:
Bad hotend and bed temperature update on status screen.
If I turn on the hotend heating and the bed turns off after reaching the temperature, the temperature on the status screen is updated and it can be seen that the temperature is falling.
If I restart the printer when the temperature is reached, the printer will display a high temperature. The temperature of the hotend and the bed drops, but the display still shows the high temperature it read after the reset.

SKRV1.4 + TFT35E3V3.0
Marlin Bugfix 2.0 from today
TFT own compilation from today:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/46d7434d8b463f160130543a2c5751aaeff1fd0c

radek8, 2020-12-25:
The second option is to simulate this problem.
Marlin has a new setup. Can set the heating to switch on automatically when leveling starts (G29)

[image]

I will enable the preheat function before leveling in Marlin
I start leveling. The bed is heated and leveled in the bed. After leveling, the heating switches off automatically.
I jump to the status screen, but the status screen does not respond to a falling temperature and still shows a high preheat temperature.

kisslorand, 2020-12-25:
For some reason your Marlin is not auto-reporting temperatures. Can you check this hypothesis?

radek8, 2020-12-25:
There will be no problem with that
this is a record of communication on the serial line,
the hot end temperature is reported at 29 ° C but the display currently shows 36 ° C

kisslorand, 2020-12-25:
Is your nozzle other than T0?

radek8, 2020-12-25:
If I open the menu and return, the temperature is still bad.
if I open the heating menu, the temperature value is updated after 1 s.
after returning to the status screen, the temperature is updated correctly.

To update the wallpaper, I need to open the heating menu.

radek8, 2020-12-25:
the same behavior is in bed and I don't have 2 beds :-)

kisslorand, 2020-12-25:
OK, than the culprit is found.
Need to send M155 Sxx

radek8, 2020-12-25:
I can already see the difference

radek8, 2020-12-25:
printer response after restart

[image]

printer response after opening the heat menu

[image]

kisslorand, 2020-12-25:
After you restart the printer, it looses the M155 Sxx command. So there's a need to send it again

radek8, 2020-12-25:
The TFT cannot analyze the temperature message line if it contains the word OK

kisslorand, 2020-12-25:
it can:
"      if(ack_seen("ok"))
      {
        infoHost.wait = false;
      }
      // parse temperature
      if((ack_seen("@") && ack_seen("T:")) || ack_seen("T0:"))
      {
        heatSetCurrentTemp(NOZZLE0, ack_value()+0.5f);
        if(!heatGetSendWaiting(NOZZLE0)) {
          heatSyncTargetTemp(NOZZLE0, ack_second_value()+0.5f);
        }
        for(uint8_t i = 0; i < MAX_HEATER_COUNT; i++)
        {
          if(!heaterIsValid(i)) continue;
          if(ack_seen(heaterID[i]))
          {
            heatSetCurrentTemp(i, ack_value()+0.5f);
            if(!heatGetSendWaiting(i)) {
              heatSyncTargetTemp(i, ack_second_value()+0.5f);
            }
          }
        }"

radek8, 2020-12-25:
The status screen cannot process the m105 command
If heating is started, M155 is started,
if I reset the printer, the M155 command is executed from the Heat.c menu, or after switching on the heating again

[image]

[image]

radek8, 2020-12-25:
If the TFT cannot analyze the M105 then it is unnecessary to send it and burden the communication if the M155 is used
using both in parallel is unnecessary.

kisslorand, 2020-12-25:
M105 is sent only if infoMachineSettings.autoReportTemp is false, So, as I said, M155 Sxx must be sent again ig printer was disconnected and reconnected.
My latest PRs burned me out so I leave it to fellow colleagues to deal with it if they have the time.

radek8, 2020-12-25:
After the printer restarts, find the temperature, but no longer update it.
The problem is either in the analysis of the response to the M105, and everyone will have it, or the format of the response has changed in the new version of Marlin.

Can you see in what format your Marlin corresponds to M105?

kisslorand, 2020-12-25:
Not tonight (1:48AM here), maybe tomorrow.

radek8, 2020-12-25:
0:49, I go to sleep :-)

kisslorand, 2020-12-25:
Send: M105
Recv: ok T:20.31 /0.000 B:20.31 /0.000 @:0 B@:0

radek8, 2020-12-26:
The repair works, thank you.
@guruathwal, not a bug in ParseACK.c yet?

[image]

should not be correct wording:
  if (ack_seen ("@") && (ack_seen ("T:") || ack_seen ("T0:")))

guruathwal, 2020-12-26:
@radek8  it is correct.  some marlin versions report temperatures with `T0:` when there are more than 1 hotends. marlin also reports heater power with `@` which is being double-checked along with `T:` to ensure that the received string is a temperature update.

radek8, 2020-12-26:
@guruathwal 
It is currently
If  (( a  && b) || c )
I think it should be
If  (a && ( b || c ))

guruathwal, 2020-12-27:
@radek8 It is not necessary that `@` is always included in the temperature string. It was added to make it compatible with Repetier firmware because without it other responses with `T:` were getting parsed as temperature. This was changed in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/959. Earlier it was like this: 
```C
if(ack_seen("ok T:") || ack_seen("T0:"))
```

[image]

radek8, 2020-12-27:
OK
Thank you.

PR #1438 (2020-12-29): Fix: temperature auto-report timeout check, Fix invalid settings parameters
- Fix for temperature auto-report disabling due to Mainboard reset or through any other serial controller (#1434). Now it checks for a timeout after last reported temperature status and resends `M155` command.
- Update old settings signature causing invalid parameter reading due to changes in parameters via PR #1359 (related detail in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1433#issuecomment-751341268)
- Update config.ini
- Fix leveling feed-rate for probe height/offset control
- Clean cod [... cut]
closed 2020-12-29 by bigtreetech