bigtreetech/BIGTREETECH-TouchScreenFirmware #1435: [BUG] Does not change FR and Flow during printing

karabas2011, 2020-12-25:
### Description

When printing I can change framerate only once and cannot return to default or further change.

### Steps to reproduce

1. Start printing
2. Go to tuning menu 
3. change frame rate
4. wait
5 change frame rate.
6 repeate 4 and 5

**Expected behavior**
changing every time

**Actual behavior**
changing only once

I can easyly change FR via printerface during this printing.

kisslorand, 2020-12-25:
There's been a bug sneaked in during the 24.XII.2020 merge, Check new master.

oldman4U, 2020-12-29:
Any news regarding the reported issue?

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2021-01-02:
karabas2011

Are you talking about printing using Touchscreen or Marlin emulation?

Please try latest firmware where it works perfectly fine at least on my printers.

kisslorand, 2021-01-02:
This was fixed by PR #1431.

sarvenn, 2021-01-10:
@kisslorand Does the latest TFT35 e3 prebuild firmware include the fix? Or should I compile by myself from the master file?

kisslorand, 2021-01-10:
It should contain it.

sarvenn, 2021-01-10:
@kisslorand I will try today. Thnx.

sarvenn, 2021-01-12:
I think it is solved now. I installed the latest prebuild firmware and did not encounter this issue. I didn't print a lot of thing but could not replicate the old issue.

@karabas2011 have you tried the latest firmware?

closed 2021-01-17 by karabas2011