bigtreetech/BIGTREETECH-TouchScreenFirmware #1439: BTT TFT35 V2.0 firmware not mentioned in the list of files - which one should I use for the screen?

Feppofap, 2020-12-26:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

kisslorand, 2020-12-26:
If you compile your own, there is an option for BTT TFT V2.0.

radek8, 2020-12-26:
Use this one
[BIQU_TFT35_APP1_V2.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5743656/BIQU_TFT35_APP1_V2.0.26.x.zip)

radek8, 2020-12-26:
[image]

Feppofap, 2020-12-27:
OK - thanks. I didn't think that was the righrt one, since irt says APP in the filename :)

closed 2020-12-27 by Feppofap