bigtreetech/BIGTREETECH-TouchScreenFirmware #1440: Z offset xy_offset_probing in config.ini ignored.

jimhiggs, 2020-12-27:
TFT35_V2.0.26 
When using the Bed Levelling Zoffset button  with XY_Offset_Probing set to 0 in config.ini the result was the same as when set to 1. I tried to find out why and could not find an entry to set the infoSetting variable in config.c.
I added the following code to config.c and this appears to fix the problem.
`    case C_INDEX_XY_OFFSET_PROBING:
      SET_VALID_INT_VALUE(infoSettings.xy_offset_probing, 0 ,2);
      break;`
I am also finding that setting baudrate in config.ini does not work, I need to go to the connection screen to set it manually. Not worked out why yet.

jimhiggs, 2020-12-27:
ok So I looked at ProbeOffsetControl.c and see that it homes first in probeOffsetEnable(bool skipZOffset). I added  mustStoreCmd("G1 X147 Y144\n"); into the code to get mine to zoffset in the center. Obviously this is only right for me but I suppose that could be a init setting in config.ini, ie position to zoffset if not at xy-offset_probing position.
I am confused though with the section of code that sets the position of z and sets the Z offset value (before you change it).  It reads:
`if (!skipZOffset)
  {
    probeHeightStart(0.0f);  // lower nozzle to absolute Z0 point
  }
  else
  {
    probeHeightStart(-probeOffsetGetValue());  // lower nozzle to probing Z0 point
    probeOffsetSetValue(0.0f);                 // reset Z offset in order probing Z0 matches absolute Z0 point
  }`
skipZOffset is true (when you click the Zoffset button to on)
So it is doing the second bit which seems to me to move the z axis to the currently set z offset point and then sets the probeOffsetValue to 0. I don't understand why it is setting it to zero. Perhaps someone can explain?

oldman4U, 2020-12-27:
There are several issues related to config.INI. Check the other tickets from yesterday and the PR section.

oldman4U, 2020-12-27:
#1438

jimhiggs, 2020-12-27:
Thanks, I searched but missed that.
Still Have the query regarding the setting of probeHeightStart in :
`if (!skipZOffset) { probeHeightStart(0.0f); // lower nozzle to absolute Z0 point } 
else 
{ probeHeightStart(-probeOffsetGetValue()); 
// lower nozzle to probing Z0 point probeOffsetSetValue(0.0f); 
// reset Z offset in order probing Z0 matches absolute Z0 point }`

I spent hours trying to set my z offset height getting nowhere, I finally got it when I set z offset to zero and saved and then readjusted for my actual value. I am not convinced the else clause is correct.
I did not find this mentioned in #1438

digant73, 2020-12-28:
the current version simply homes and moves the nozzle to probing Z0 (this means Z offset is subtracted).
In case xy_offset_probing (in config.ini) is set to 1, after homing, XY offset are applied (Z offset continues to be subtracted)

jimhiggs, 2020-12-29:
OK Thanks for the update, I thought it was moving z to the previously set Z but realise now it was going to actual z0.

closed 2020-12-29 by jimhiggs