bigtreetech/BIGTREETECH-TouchScreenFirmware #1442: [BUG] Freezing TFT35 E3 V3

Bastlwastl84, 2020-12-28:
Freezing 
Power on
Preheat Nozzle PETG
also no Update of Temperature
Move
Messange Please wait ant non responsing
[config.ini.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5746771/config.ini.txt)
resetted yesterday with reset.txt
firmware from yesterday
icons, font and languagefile from yesterday
https://youtu.be/bPxHzax0d1M

kisslorand, 2020-12-28:
This reminder message appears only when the command queue is full.
Just checked and cannot reproduce the issue, no background process fills up the command queue.
Can you check the communication between TFT and motherboard after you press preheat?

Bastlwastl84, 2020-12-28:
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Bastlwastl84, 2020-12-28:
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Bastlwastl84, 2020-12-28:
Couldn't it be done like before, if you had edited your configuration.h andxself compiled, than you don't need cinfig.bin

kisslorand, 2020-12-28:
It seems you do not have the latest version where the memory allocated for language file is increased by 1KB.

Bastlwastl84, 2020-12-28:
I downloaded it today again and than that many errors. I did inserted all settings identically. After that i was frustrated and installed the tft Firmware again from June.

kisslorand, 2020-12-28:
[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5747984/BIGTREE_TFT35_V3.0_E3.26.x.zip)

Put these on your SD Card and edit config.ini to your needs. Pay attention to the baudrate, you have to write the baudrate itself, not its index in the list.

Bastlwastl84, 2020-12-28:
I can try tomorrow. Printing now in Marlin Mode.

oldman4U, 2020-12-28:
This user believes that he does not need config.ini even several users already told him.

Bastlwastl84, 2020-12-28:
> This user believes that he does not need config.ini even several users already told him.

my old bin had the settings i selected at configuration.h?

kisslorand, 2020-12-28:
ZIP file reuploaded, download it again.

Bastlwastl84, 2020-12-29:
do you also have the german language file to integrate, too, when testing tomorrow?

Bastlwastl84, 2020-12-29:
reset.txt, than powered off, reboot with this firmware and config.... only 1 error, cause of hotend maximum Temperature

Bastlwastl84, 2020-12-29:
[btttft35e3v3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5751746/btttft35e3v3.zip)

Bastlwastl84, 2020-12-29:
Connected with Pronterface after that: While the TFT35 is not responsive i could controle the Printer with that tool. Plugged in esp Module. Responsive ESP01 Module, while not Responsive TFT. Same with ESP32

kisslorand, 2020-12-29:
So the TFT freezes only if you go to preheat and after press Move? No other scenario leads to freezing?

Bastlwastl84, 2020-12-29:
> So the TFT freezes only if you go to preheat and after press Move? No other scenario leads to freezing?
> Do you have M155 enabled in Marlin?

my marlinconfig is under last picture

Bastlwastl84, 2020-12-29:
i could try more tomorrow (printing in marlin mode now)

Bastlwastl84, 2020-12-29:
it must be a serial communication problem. 

This Evening I plugged in, before this print in Marlin Mode, the ESP32-Cam, Now I lost communication from ESP32-Cam to Board again. I changed the cable to TFT with a old USB-Cable, i crimped, but nothing changed. 

The ESP32-Cam is Plugged in at RX3 and TX3 at TFT35. I asked Luc about Serial at ESP3D config. #define ESP_SERIAL_OUTPUT USE_SERIAL_0 
Luc said "yes if you use SERIAL0
which is the default one - same as usb"

kisslorand, 2020-12-29:
Glad you find a possible issue!

Bastlwastl84, 2020-12-29:
> Glad you find a possible issue!

perhaps i'll try Marlin 2.0.5.x to see, if there's the bug, too. 
but now printing Parts for a Veron 2.4

Bastlwastl84, 2021-01-01:
[image]
must be a Serial Problem, but where it comes from?

Bastlwastl84, 2021-01-01:
at TFT 5V is 4,8V, Could that be the Problem?. Where to adjust?

Dieter59-GIT, 2021-01-03:
I have the same problem and was able to find out some things.
I can confirm the assumption for a serial port problem.
My supply voltage is also 4.8V, but this is not a problem. The TFT35 board still works in my tests with about 4V.
I found out that the supply voltage of the printer controller board has a big influence on the occurrence of the error.
My board: SKR1.4 Turbo
Power supply: 24V (with quite strong disturbances when switching on the hotbed)
When printing, disturbances in the serial transmission occur several times during printing (wrong Gcode and wrong traverses) or the printer freeze sometimes. 
My attempts:
1. External supply of the SKR1.4 with a 11.4V-Lipo-battery -> print runs without errors
2. External supply of the SKR1.4 with two in series connected lipo-batteries (22.8V) -> print has failures
3. Internal 24v-supply via DC/DC-converter reduced to 13V and SKR1.4 operated with it -> print runs without errors

No. 3 (a DC/DC converter with filter element) is my solution at the moment.

Bastlwastl84, 2021-01-03:
Could be a buffermodule, like the BTT UPS 24V (https://www.biqu.equipment/collections/all-product/products/btt-ups-24v-v1-0-resume-printing-while-power-off-module), bring the solution?

Dieter59-GIT, 2021-01-03:
My hotend and hotbed is also controlled by external mosfet.
The disturbances on my 24V supply appear at high current load (narrow pulses with approx. f=140kHz). 
Whether the modules brings an improvement I can not say.
I have tried filter circuits of capacitors and inductors, but the problem always came back. Only the reduction of the 24V supply voltage to approx. 12V made error-free prints possible.
An adjustable DC/DC converter with 3-5A is sufficient for the reduction. 
At this point, however, I can't say what exactly is causing the communication problem.
It looks like synchronization is lost when sending char. This causes the sent data set to be garbled. If it happens in the numeric area, the print head goes strange ways, if it happens at the position of the G-command, an error message "Unknown Gcode ..." appears.  
I realize that this does not fix the cause but only the effect. 
However, I will continue to observe the behavior. 
Maybe it can be fixed by software (e.g. more stable RS232 communication).

Bastlwastl84, 2021-01-03:
When powering the TFT35 i got many fast short beeps and at Display after Boot this messages. Correctly no SD Card is at Mainboard.
[image]

Bastlwastl84, 2021-01-03:
[image]
[image]

Bastlwastl84, 2021-01-05:
Again not updating
https://youtu.be/4fVjKl23IGE

Bastlwastl84, 2021-01-05:
perhaps that? https://github.com/MarlinFirmware/Marlin/issues/20673

Bastlwastl84, 2021-01-09:
I updated now again, freezing when opening Extruder.

closed 2021-03-18 by stale[bot]