bigtreetech/BIGTREETECH-TouchScreenFirmware #1443: How select the correct .bin?

nachogil, 2020-12-28:
Hi,

I'm having some difficulties to update my Bigtreetech TFT35 V3.

Basically there are those .bin options:

- BIGTREE_TFT35_V3.0_B1.26.x.bin
- BIGTREE_TFT35_V3.0_E3.26.x.bin
- BIGTREE_TFT35_V3.0.26.x.bin

But I'm not sure which .bin I must use
This is is my TFT

[image]

kisslorand, 2020-12-28:
It seems "BIGTREE_TFT35_V3.0.26.x.bin" would be the one for your TFT.

oldman4U, 2020-12-28:
Hi.

According to the attached screenshot, you own a BIGTREETECH TFT35 v3, so firmware BIGTREE_TFT35_V3.0.26.x.bin is the correct variant.

oldman4U, 2020-12-31:
Nine day wonder.

Answer or close this ticket.

closed 2021-03-08 by stale[bot]