bigtreetech/BIGTREETECH-TouchScreenFirmware #1444: max temperature of heat for noozle

daniagos, 2020-12-28:
tft 35 v2.0
i can't make the max temperature of the nozzle, it seem fixed to 275

i changed in marlin max temp to 550 degrees, changed also in tft configuration.h and in config.ini to 550, uploaded the firmware but still max temp allowed from tft is 550.

if i set the temperature to from pronterface i can go as high as 550 set in the board, and tft display  the correct temperature.

Latest firmware available,

What can i check?

Thanks

kisslorand, 2020-12-28:
Check "MAX_TOOL_TEMP" in config.c.

oldman4U, 2020-12-28:
Maybe you want to try to configure the config.ini file first, where one part is the max temp:

#### Bed / Extruder / Chamber Maximum Temperatures
# Format: [max_temp: T0:<max temp> T1:<max temp> T2:<max temp> T3:<max temp> T4:<max temp> T5:<max temp> BED:<max temp> CHAMBER:<max temp>]
#
# Unit: [°C]
#
max_temp:T0:275 T1:275 T2:275 T3:275 T4:275 T5:275 BED:150 CHAMBER:60

Reading the ReadMe could eventually help further: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#configuration

oldman4U, 2020-12-30:
Any news?

Please help the community and close the ticket once you do not need it anymore. Thank you

daniagos, 2020-12-30:
Hi, i reflashed the config.ini a couple of times and now it works, i didn't do anything different, but now it works, thanks for help

closed 2020-12-30 by daniagos

oldman4U, 2020-12-30:
👍🏻

Happy printing