bigtreetech/BIGTREETECH-TouchScreenFirmware #1445: TFT35 E3 V3 keeps clicking and going back to the mode selection screen

TRiXWoN, 2020-12-28:
Hi there, have an Ender 3 with a SKR Mini E3 v2. 
Just added the TFT35 E3 V3 and on first boot it says no printer then seems to connect after a few seconds, then it starts clicking from the speaker and goes back to the mode selection screen with MARLIN/USB on it. 

It used to say Marlin/Touch but I upgraded the firmware and now it says USB. Also since the firmware upgrade some of the icons and graphics when it does shortly go into the touch mode look corrupted.

I cannot use it at all once the clicking and resetting starts. I did verify that it was set to the proper baud mode before it started resetting over and over. 

It has never worked, did this shortly after first boot.

- I verified that the serial/tft cable is plugged in correctly with the spare pin on the far left
- I tried booting with just the EXT03 cable plugged in and it does the same thing
- Tried booting with just the serial/tft cable and it does the same thing
- Tried removing it from the housing to see if it was grounding out or holding down the reset button and still did the same thing.

kisslorand, 2020-12-28:
Check this ZIP file I uploaded to another user who has also TFT E3 V3: [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5748788/BIGTREE_TFT35_V3.0_E3.26.x.zip](url)

oldman4U, 2020-12-28:
Remove the TFT from the metal frame of the Ender to see if this solves the problem.

TRiXWoN, 2020-12-28:
> Remove the TFT from the metal frame of the Ender to see if this solves the problem.

Already stated I did that no change.

oldman4U, 2020-12-28:
You wrote: It has never worked, did this shortly after first boot.

An issue like this out of the box = most likely a hardware issue.

Background information. As soon as the TFT gets power, you will see the touchscreen and you can navigate through the menus, even no RS232 or EXP cable is connected. You can do this by cutting an old usb cable and connect the red and black wire to 5V and GND of the TFT. You will get a No printer attached in this case, because the TFT can not see a mainboard via RS232. You will be able to switch to Marlin mode, where you get a single line of text at the top. Marlin mode would be visible as soon as the TFT can see Marlin through the EXP connector.

Hope this helps

TRiXWoN, 2020-12-28:
> You wrote: It has never worked, did this shortly after first boot.
>
> An issue like this out of the box = most likely a hardware issue.
>
> Background information. As soon as the TFT gets power, you will [...]

Thanks I will try powering it with no printer and see what happens

oldman4U, 2020-12-28:
And maybe you can attach a movie here which shows the problem.

TRiXWoN, 2020-12-29:
> And maybe you can attach a movie here which shows the problem.

I added video. I also re-updated firmware and the graphic corruption issues I was having are gone. It still just clicks and goes back to the main screen over and over. My stock ender3 display from creality works fine.

rightawy, 2020-12-29:
This is a hardware issue, most likely the edges of the touchscreen are permanently stuck together creating continuous long-press which is causing the TFT to show the mode select screen. Try rubbing your fingernail gently on all four edges of the touchscreen, if this does not help please contact BTT or the store where you purchased the TFT from.

radek8, 2020-12-29:
Save  in SD  card Reset.txt

oldman4U, 2020-12-29:
Based on your movie this is hardware related. You can try to fiddle around with the encoder or the edges of the screen. But I would contact the dealer you bought it from.

Please help the community and close the ticket once you do not need it anymore. Thank you

radek8, 2020-12-29:
If the calibration of the touch layer does not help, it will be a HW problem

TRiXWoN, 2020-12-29:
> Save in SD card Reset.txt

After the calibration (which took about 4 tries it kept saying it couldn't be completed, looked as if it was registering multiple touches per touch) if I only have the tft cable plugged in then it worked for a print but went back to the mode selection again.

I choose touch mode and it stayed there for a couple hours. 

Turned printer off, connected the marlin mode cable to ext3 and the ticking started again and it did the reset to menu issue again.

Disconnected the marlin mode ribbon cable and did reset.txt again and the issue is still happening. 

I'm going to try and exchange it. My old ender3 display works perfectly fine, I just wanted the functions of being able to enter marlin commands, direct filament runout sensor connection, and maybe even the wifi later. If the second device comes and has the same issue I'll just go back to the old screen.

oldman4U, 2020-12-30:
You can also disconnect ALL cables from the TFT and connect only a "modified" USB cable. For this, you have to cut a USB cable and connect the red and black wire to 5V and GND of the TFT. This will allow you to update, reset and test the TFT independent of a mainboard and Marlin.

TRiXWoN, 2020-12-30:
> You can also disconnect ALL cables from the TFT and connect only a "modified" USB cable. For this, you have to cut a USB cable and connect the red and black wire to 5V and GND of the TFT. This will allow you to update, reset and test the TFT independent of a mainboard and Marlin.

You just jam them in the connector or do you have some better way to keep the connection reliable aside from soldering or crimping ends on a cable for this one time test?

oldman4U, 2020-12-30:
I crimped a connector for this and would recommend to do the same. It takes
less time than fiddling around.

TRiXWoN, 2020-12-31:
> I crimped a connector for this and would recommend to do the same. It takes less time than fiddling around. TRiXWoN <notifications@github.com> schrieb am Mi. 30. Dez. 2020 um 23:03:
> […](#)

Ok I got the replacement in wired up with usb cable and out of the box looked like this 
[image]
I updated it and it looks ok now have not seen it resetting or clicking. Going to test the old one too just to see.

oldman4U, 2020-12-31:
How does it look now and do you still have the clicks? Can you navigate
through the menus?

TRiXWoN, 2020-12-31:
Tested the original LCD with the issue by using the USB cable and it did the same thing it did when connected to the printer. Clicking and resetting back to mode selection.

TRiXWoN, 2020-12-31:
> […](#)

It looks good when connected to computer with the USB cable after updating firmware. I can navigate menus and use the knob without issue. No clicking, no resetting. Not sure why it was so messed up out of the box.

The first one I had still reset and could not be used when just connected with USB cable so I'm pretty sure it was defective. Going to test this new one on the printer and then close out this issue if its fixed.

oldman4U, 2020-12-31:
> in wired up with usb cable and out of the box looked like this [image:
> 82F7012B-3930-4A46-983B-C2CB4AAAB0B3]

Sounds good.

Most likely cables are the issue and less likely the TFT. But we also saw
DOA hardware here.

👍🏻

radek8, 2020-12-31:
> It looks good when connected to computer with the USB cable after updating firmware. I can navigate menus and use the knob without issue. No clicking, no resetting. Not sure why it was so messed up [...]

From what you're writing, it looks like you connected the display through the wrong connectors. Can you send a photo as you are connected?

TRiXWoN, 2020-12-31:
> From what you're writing, it looks like you connected the display through the wrong connectors. Can you send a photo as you are connected?

For this test it was just power and ground via USB to the +5v and GND on the TFT/rs232 connector just to make sure the device worked.

When the original display was connected to printer it was connected via the supplied cables to tft/rs232>TFT connector on SKR Mini E3 v2, and the ribbon cable to EXT3. This new display does not have the same issues. I don't know why the shipped firmware had corrupted icons but after the firmware update it looks fine.

oldman4U, 2020-12-31:
So the TFT works but the real life test with the mainboard connected is pending.

Fingers crossed;-)

TRiXWoN, 2021-01-02:
> Any news?

Not yet sorry had some things I needed to print and didn’t want to risk it not working

oldman4U, 2021-01-02:
Please let us know as soon as you have time to check.

Thank you

TRiXWoN, 2021-01-03:
It appears to be working. Still need to print with it but it’s staying on 
[image]

closed 2021-01-03 by TRiXWoN

oldman4U, 2021-01-03:
Cool.

Happy printing