bigtreetech/BIGTREETECH-TouchScreenFirmware #1449: [BUG] Wrong filament length reported if flowrate is increased or decreased

rightawy, 2020-12-29:
### Description
The filament length reported is used from hardcoded values in the gcode file. This is only true if no change is done to printing settings during print. When the flow rate is increased or decreased during printing, this change the actual filament used but still it report the old hardcoded filament length. this is really not good when printing a large model where some times the flowrate needs to be adjusted based on the actual appearance of the printing. for a large print which can take more than 6 hours to complete, changing the flow rate even for a few hours will hugely affect the final filament length.

### Steps to reproduce

1. slice a large print which takes a lot of time to print or a lot of filament to complete
2. change flowrate by 10%  during the print for half time.
3. it still shows old length at the end.

**Expected behavior**
show actual length of the filament used.

**Actual behavior**
shows old filament length used regardless of increasing or decreasing flow rate.

### Hardware Variant
TFT35 V2
### TFT Firmware Version & Main Board Firmware details
 latest firmware dated 28 dec 2020.

kisslorand, 2020-12-29:
It has been written that the flow rate at the end of print is taken into consideration. Also it has been written that, as it is the code right now, if the flow rate changes during print (and not at the beginning) the results will be inaccurate.
 PR #1359 :
"Filament data takes into account the flow rate also but with a caveat. It has to be the same flow rate during the entire time of the printing, because the end result is calculated based on the flow rate at the time the print has finished. If flow rate changes during the print the results will not be accurate anymore."

Readme should be updated to reflect this.

rightawy, 2021-01-01:
> I mentioned it before, flow rate should be adjusted on the TFT only for calibration purposes. In the whole world the standard is to make calibrations/adjustments at input, not output. Of course, this is not the case every time, shortcuts are made, but that's another story...

If you believe it, then you should remove the flow rate option during printing and move it to the machine tuning menu. This would be perfect in a perfect world only. There will always be some prints which will need some tuning in flow rate and feed rate. Many times to achieve the desired strength the flow rate needs to be increased, or when some walls do not merge together when everything was perfect during the calibration prints.  if the values are not correct when the flow rate was changed then why show it after flow rate change?

kisslorand, 2021-01-01:
Ticket can be closed, it is a documented behaviour, not a bug.
There is always room for improvements, users can suggest or ask for them in a the form of feature request.

oldman4U, 2021-01-01:
Besides some other information, this will be added to the next release of the ReadMe:
Calculations are done in both absolute or relative extrusion mode. Filament data takes into account the flow rate also but with a caveat. It has to be the same flow rate during the entire time of the printing, because the end result is calculated based on the flow rate at the time the print has finished. If flow rate changes during the print the results will not be accurate anymore.

Even it is not the perfect solution, it will work for most of the prints and does not harm the quality of the print by slowing down communication or overfilling a buffer. Users (like me) changing the Flow during a print have to be aware, that the value shown at the end of a print can not be accurate.

oldman4U, 2021-01-02:
rightawy. Would you like to open a FR for this?

rightawy, 2021-03-04:
I have now just disabled the feature. It is better to have no info instead of wrong info.

closed 2021-03-04 by rightawy