bigtreetech/BIGTREETECH-TouchScreenFirmware #1451: [BUG] (short description)

DanJunior78, 2020-12-29:
### Description

with latest project from 29.12.2020:
First i am starting in TouchUI mode. All is working. I switch to MarlinUI and also all is working properly. If i switch back to TouchUI it´s not reconnecting to the mainboard.
I´m using a new set of mainboard (BTT SKR Pro V1.2 with latest Marlin 2 bugfix from 20.12.2020) and BTT TFT 70 V3.0 with all files from 29.12.2020

### Steps to reproduce
1. Start printer in TouchUI mode
2. Switch to MarlinUI
3. Switch back to TouchUI

**Expected behavior**
Reconnect to the mainboard after switch back

**Actual behavior**
on top of the screen "please wait..." after trying to move axis stays active. Board and TFT keep responsive. So, no TFT "software hanging"

### Hardware Variant
BTT SKR Pro V1.2 with latest Marlin 2 bugfix from 20.12.2020) and BTT TFT 70 V3.0
### TFT Firmware Version & Main Board Firmware details
BTT SKR Pro V1.2 with latest Marlin 2 bugfix from 20.12.2020) and BTT TFT 70 V3.0

### Additional Information
Attached my config file
[config.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5750541/config.txt)

I´ve tested following mistakenly: serial_always_on. i´ve tried both modes. With the serial always on (=1) it was reading but not able to send commands in TouchUI mode. In Marlin mode was everything fine.
With the serial_always_on = 0 it is working with the above problem.

DanJunior78, 2020-12-29:
Some news...i can resolve this problem by going to the settings and disconnecting the screen in settings-conncection-disconnect.
But it´s not stable for commands. Sometimes it´s directly moving for manual axis movement after seconds of disconnecting it...but sometimes not.

DanJunior78, 2020-12-29:
No, after switch on and off, it´s not reconnecting. It´s more like to have good look in Touch UI mode. MarlinUI is flawless working.

oldman4U, 2020-12-29:
Can you check if there is a baudrate activated under Connection in case the connection is lost.

DanJunior78, 2020-12-29:
Yes, the baudrate has the green ✅

DanJunior78, 2020-12-29:
And it's really like... disconnect via button and after a few seconds or 2 or 3 disconnects that it pushes the Gcodes to the mainboard...the temperature values stay alive...ok, at room temperature.

oldman4U, 2020-12-29:
Have you tried an older firmware? From a month ago for example.

DanJunior78, 2020-12-29:
Of the TFT?

oldman4U, 2020-12-29:
Yes.

DanJunior78, 2020-12-29:
i had one from beginning of December. It was from end of November (self compiled). There was some bugs, e.g. z axis in MarlinUI was moving when going into the z axis movement...it started to move upward without stopping. So, i decided to use this new one. Which is, at least in MarlinUI, working fine in what i´ve used it till now (getting the printer up and running....it´s still ongoing).

Funny thing...i have now switched back to TouchUI during heating up and it was showing the correct temps and...i could move z axis. I don´t know what it could be...but...i am hard in trying ;).

I´ve still another combo with a BTT SKR Pro v1.2, a BTT 1.4 Turbo and a BTT TFT 50. The last two i´ve had setup and running on the desk too, but still not assembled in a printer.

DanJunior78, 2020-12-29:
When i go to Parameters -> Printer and it was on state "please wait...". The display gets stucked and it is not possible to navigate again, it means...it is not responding anymore.

oldman4U, 2020-12-29:
I have a SKR PRO v1.1 with Marlin 2.0.7.2 and TFT 43 with firmware from yesterday and it works fine. What happened between working well and now? Only TFT firmware? If so flash an older fw like described on the main page.

DanJunior78, 2020-12-29:
Nope, I've exchanged mainboard and TFT on my stock Anycubic Mega X. The mainboard was tested offline only and sure the TFT too. All these boards are not older then 5 weeks. I had no time to exchange it before. Before I've used MKS Boards and TFT for some printers. In the last months I've replaced 6 boards and screens on different printers. I can't tell about older versions of BTT products. Only that with the November version my z axis moved endless to the top in Marlin UI when trying to get to z axis movement. In this latest version it's working fine, it's going like for x and y to the selection of distances and further.
So, I've no clue what could be wrong.
When I push the reset button on the screen the reset on the mainboard is initiated. Sometimes TFT reconnects, sometimes not. 
Usually I'm not touching things so extremely,because it's not required when I'm printing. But during setting up the system, it's like lifetime usage of the screen,lol.

DanJunior78, 2020-12-29:
I'm using latest Bugfix Version of Marlin,cause I had used it to get a MKS System running which is still not on the 2.0.7 available. 
Is it possible to debug the serial interface? I didn't check till now the hardware in detail. I'm open for all...even for programming support at bug fixing. Even I'm right now heavy busy with my GitHub projects.

oldman4U, 2020-12-29:
This is all much to complicated. Usually you connect it, set the serial port and baud rate and the right kind of screen, like described - and it works.

You said you replaced other boards. Same type? Do they work correctly?

DanJunior78, 2020-12-29:
no. It´s the first BTT board and TFT. I´ve planned to exchange another printer with nearly same setup only with the TFT50.

oldman4U, 2020-12-29:
You could try to cut a USB cable and connect the red and black wire to 5V and GND of the TFT. Remove all other cables from the display before. Check if you can switch between the modes and if you can navigate through all the menus. Of course doing so you get an empty Marlin screen and a No printer connected error in touchscreen.

Let me know

DanJunior78, 2020-12-29:
I´ll try to connect the 5V at the power inlet. I´ve all electronic bits and parts at home and come back with more. Thanks for your really fast feedback and support (in between our "conversation" :o) )

DanJunior78, 2020-12-30:
Ok, so, i believe, i know what is the problem. Sorry for bothering you for so long time and i really appreciate your really good support.
I tested with 5V as you said. This was not changing anything, because i think the problem is coming from the mainboard.
I did some tests last night. I even flashed Marlin 2.0.7.2. Finally i was curious and i took out the new BTT SKR Pro V1.2 board and the TFT50.
Both are usually connecting to the mainboard on first start up. Only, on this one installed. When i press the reset button on the TFT. The mainboard seems to stay frozen and not booting up again. The new board does not have this effect after pushing reset. Both displays reconnect on the new board nice. Only on the installed mainboard something seems to be wrong.
Good, anyways. I wish you a happy new year 2021...and keep going with this nice project.

DanJunior78, 2020-12-30:
I am closing it...

closed 2020-12-30 by DanJunior78

oldman4U, 2020-12-30:
👍🏻

Happy printing