bigtreetech/BIGTREETECH-TouchScreenFirmware #1453: [FR] (feature request title)

kisslorand, 2020-12-29:
Someone would be so kind to include the following informations in Readme.md? 

----------------------------------------------------------------------------------------------------------------------------------------------
Statistics after the print is finished:

Statistics as filament length, filament weight and filament cost can be embedded into the gCode. These statistics will be presented after the print is finished in the new infobox area as a scrolling text. You can click that infobox and a popup will present you the printed filename (limited to the first 25 characters), the time needed for the print, the filament length used, the filament weight and its cost.

The statistic data in the gCode must have the following format*:

M118 P0 filament_data L:{12.3456}m
M118 P0 filemant_data W:{1.23456}g
M118 P0 filament_data C:{0.1234}

*a good practice would be to include this at the beginning of the gCode.

L: represents the length in meters, W: represents the weight in grams, C: represents the cost without a unit.
The values of every filament data can be in a brackets, parentheses, apostrophes, etc. or without them, measurement units can be there or not.
So M118 P0 filament_data L:(12.3456)m, M118 P0 filament_data L:12.3456meters, M118 P0 filament_data L:[12.3456] and so on are all valid formats.

The inclusion of the filament data into the gCode can be automated. In Cura all you have to do is to insert the following into the Start G-Code:
M118 P0 filament_data L:{filament_amount}m
M118 P0 filament_data W:{filament_weight}g
M118 P0 filament_data C:{filament_cost}

This way you will always have filamant data in your gCode files automatically, without further user intervention.

In case filament data is not present in the gCode, the filament length data is calculated during print. Upon testing the measurements were quite exact, they were under 1mm precision. Length is calculated regardless of using the TFT USB, TFT SD or the onboard SD. Calculations are done in both absolute or relative extrusion mode.
The MCU usage is increased by a tiny fraction by these calculations, no influence on print quality could be detected during tests.

Filament data takes into account the flow rate also but with a caveat. It has to be the same flow rate during the entire time of the printing, because the end result is calculated based on the flow rate at the time the print has finished. If flow rate changes during the print the results will not be accurate anymore.

----------------------------------------------------------------------------------------------------------------------------------------------

oldman4U, 2020-12-29:
Just pushed a PR for an update of the ReadMe, but because we are such a great team, I set the PR on Draft and will change the ReadMe.

Not sure you want to hear my opinion, but in case....

It is my feeling that the average user does not read even the most important things. Your text above is much too long and it would be good to shorten the text and maybe even to split it and add some information to the readme and some to the configuration files.

Just let me know where you would like to place which text exactly.

kisslorand, 2020-12-29:
I couldn't agree more than 100% with you... I know, people do not read... I really do not know how to split it to be sure 1% will ever read it and not open a ticket.
To be honest, I thought including it to Readme.md as having a place to redirect tickets about filament usage reports. Anything included in the config.ini will be more buried. I really do not know what would be the best choice.

oldman4U, 2020-12-30:
There is a Wiki. Not sure how we can activate it.

Codel1417, 2020-12-30:
this sounds like it should also be added to the btt 3d plugin for cura to automatically add the gcode to a file

kisslorand, 2020-12-31:
There are 2 sentences that can be deleted, the rest is important to understand the usage of the print statistics feature. I guess it doesn't fall into the "shortening" criteria...

oldman4U, 2020-12-31:
Whats about @Codel1417 idea to add it to the plug in?

Which 2 sentences?

Codel1417, 2020-12-31:
@oldman4U im going to attempt to add it tomorrow if cura has an api to retrieve the data.

kisslorand, 2020-12-31:
@oldman4U 

> Which 2 sentences?

1. "Upon testing the measurements were quite exact, they were under 1mm precision."
2. "The MCU usage is increased by a tiny fraction by these calculations, no influence on print quality could be detected during tests."

Codel1417, 2020-12-31:
I opened a pull request https://github.com/bigtreetech/Bigtree3DPluginSuit/pull/6 for adding the material info.

oldman4U, 2020-12-31:
The BTT plug in is for generating the 3D model preview which is exclusively available for Cura.

oldman4U, 2021-01-01:
Hi Codel1417

Should I mention the Cura Plug In also. Do you have a short text for me to add to the readme?

oldman4U, 2021-01-01:
Added info about the BTT Suit. Please check #1455 and let me know

kisslorand, 2021-01-01:
Looks sweat. Thank you.

oldman4U, 2021-01-02:
Perfect.

You can close this ticket if you want.

closed 2021-01-02 by kisslorand