bigtreetech/BIGTREETECH-TouchScreenFirmware #1458: [Q] (Your Question?)where the BTT mini UPS v1.0 has to be connected to the TFT35-E3 V3.0?

killrob, 2020-12-30:
Hi, i have a TFT35 E3 V3.0 and a BTT mini UPS V1.0, where i have to connect the mini UPS to the TFT?
in UART connectors or in PS-ON connector?
`#--------------------------------------------------------------------
# Power Loss Recovery & BTT UPS Settings (if connected to TFT controller)
#--------------------------------------------------------------------
#

#### Default Power Loss Recovery Mode
# Enabled by default.
# Disable to reduce the loss of SD card or U disk
#
# Options: [enable: 1, disable: 0]
#
pl_recovery_en:1

#### Power Loss Recovery Homing
# Home before power loss recovery
#
# Options: [enable: 1, disable: 0]
#
pl_recovery_home:1

#### Power Loss Z Raise
# Raise Z axis on resume (on power loss with UPS)
#
# Unit: [mm]
#
pl_z_raise:10

#### BTT UPS Support
# Enable BTT UPS
#
# Options: [enable: 1, disable: 0]
#
btt_mini_ups:0`

this is the code in config.ini, but you don't say where the mini UPS has to be connected in TFT35
thanks

oldman4U, 2020-12-31:
Guess not many here use this hardware because resuming a print works out of the box using the TFT alone.

killrob, 2020-12-31:
> 
> 
> Guess not many here use this hardware because resuming a print works out of the box using the TFT alone.

Sure it works, but when there is a blackout the nozzle stops there and the print will be ruined, i would like to use the mini UPS to raise Z at least 10mm, but in touch mode doesn't work, i have still to try in simulator mode.

oldman4U, 2021-01-01:
It is not supported. Please see #1193

mpcolman, 2021-01-02:
That's a big issue... I have the UPS module to install it and Im breakfast with this throuble........

killrob, 2021-01-02:
well.. i think i will uninstall it until BTT release some new firmware for TFT that makes mini UPS work connected to it, as the Smart filament sensor do. I'm not intersted to use it in Marlin mode.

oldman4U, 2021-01-02:
@berrenger was able to use it with the Touchscreen but also with limitations. You have to watch the video he provided in #1193.

killrob, 2021-01-02:
i saw the video but he uses a BTT SKR V1.4, i use a mini E3V2.0. 
But no probs i will wait for BTT ;)

oldman4U, 2021-01-02:
So the problem is in the mainboard and not the TFT?

You know that the firmware is made by users like you and me, so waiting for BTT to implement it can take a long time.

killrob, 2021-01-02:
not really, the problem is that i would like to use mini UPS in touch mode with my TFT35E3V3.0 and the SKR mini E3V2.0, but i'm not so skilled in programming like our friend @berrenger, so i will wait for someone who will do the job, meanwhile i will try to take a look on source code of the TFT35 firmware to try to understand if is a job that i can do by myself.

oldman4U, 2021-01-02:
Maybe @berrenger can help. He seems to be a smart guy.

killrob, 2021-01-02:
maybe, but i will not ask cause i don't want to bore him with a problem that is mine. I will take this chance to become a little more skilled in firmware programming 😄

oldman4U, 2021-01-02:
But we help each other here and the benefit is that you meet new persons and that you learn a lot. I would also be interested how it works and once I know, we can add this information to the ReadMe. So hopefully berrenger shows up here and helps us;-)

killrob, 2021-01-03:
today i have do nothing cause a blackout for quite all day, but i agree with you, once i find the solution, if i find it, i will post here, so can be added to the readme.

berrenger, 2021-01-03:
Hi there. I stopped using "btt mini ups".  I constantly encounter different problems, so I stopped using it.  I hope you can create a synergy and solve all the problems.

oldman4U, 2021-01-03:
Thank you. This would be great.

killrob, 2021-01-04:
> 
> 
> Hi there. I stopped using "btt mini ups". I constantly encounter different problems, so I stopped using it. I hope you can create a synergy and solve all the problems.

hi @berrenger thank you for reply, are you using something else? perhaps the MKS mini ups?

gordo3di, 2021-01-05:
@killrob I've done a lot of testing with the BTT UPS 12v v1.0 module, SKR v1.4 turbo, PLR and Marlin 2.0.5.x. After exhaustive efforts here are my findings:

- BTT module must be connected to the SKR board. I used the "PWRDET" pin recommended. Its basically pin P1.0 and just turn on all the stuff in Marlin.
-
- I could not get the z-raise to work when the power loss is detected. The one time I managed to get it working (magically) the z did raise but the PLR didn't recognize so it was pointless... Still more testing to do...

- PLR works in Marlin emulation mode and it will go back to the correct position. Problem is that the hotend didn't wait to heat before moving so if it cooled on the print it will sometimes get stuck. Otherwise works correctly.

- PLR works in TFT mode but only in the traditional sense (save the recovery per layer and reprint the layer from the start). The firmware for the TFT needs to have the PLR detection implemented. On another note, the hotend does heat before moving though.

Hope this helps. I'll continue to work on it.

killrob, 2021-01-05:
good to know @berrenger, i have almost done the same things you have done, with same results, and now i have no other ideas on how go on trying.

closed 2021-03-14 by stale[bot]