bigtreetech/BIGTREETECH-TouchScreenFirmware #1460: [Q] Support MKS SGEN_L mainboards

doug-scott, 2020-12-30:
Will the BTT TFT35E3V3.0 connect to a MKS SGEN_L V2.0 motherboard and be able to work on either format (touchscreen and Marlin)?  All the online guides show having to connect the 5 pin serial cable, plus the two 10 pin flat or ribbon cables to the EXP ports on the motherboard and the TFT35 ports.  The MKS SGEN_L boards do not have the 5 pin serial cable connection.  I am unable to locate any information that suggests what to do beyond saying to connect the 5 pin cable.  I tried one but could not get past the no printer found error message.  And the touchscreen interface could not talk to the motherboard to save any changes, or to display any terminal communication between the touchscreen and the mainboard.  

Is the MKS SGEN_L v2.0 capable of working with the BTT TFT35e3v3?

FlailAway, 2020-12-30:
Yes, it will work. Well, at least with the Gen-L-v1 that I have. But you will need the nightly version of the BTT-v26 or the stable BTT-v25 version. The stable BTT-v26 will show "no printer attached."

While the "no printer..." is showing you cannot "talk" to the board. But, if you have the AUX-1 and AUX-2 cables plugged in correctly (the Geln-L-v1 has the back to front) then hold down the BTT button for 3-secs and select "Marlin." 

Wait a while (15+ secs) as it needs to boot the Marlin. Once done, you can hold down the BTT button again and go back to Touch and the "no printer" should have gone.

The Gen-L-v1 board does have the 5-pin socket but you only need +vdc, -vdc, Rx and Tx so find them on the v2.0 board. The 5th one is Reset and not needed.

doug-scott, 2020-12-30:
I can get it working half way by playing with cable arrangement..  Are you saying that the serial cable must also be attached?  I have the pins image for the mks sgen_l v2, but do not see any marked as being rx and tx.  Would they be located within the AUX1 header(pin socket)?
I have attached the image from Makerbase that shows pin numbers and locations.
[image]

radek8, 2020-12-30:
> I can get it working half way by playing with cable arrangement.. Are you saying that the serial cable must also be attached? I have the pins image for the mks sgen_l v2, but do not see any marked [...]
 
Yes, use the AUX1 connector

oldman4U, 2020-12-31:
Thank you radek8 for helping out even the topic has nothing to do with this repository.

Wouldn't it be good to add a short section to the readme related to MKS boards. Do you want to send me a short text and maybe a picture if this would help.

doug-scott, 2020-12-31:
> > I can get it working half way by playing with cable arrangement.. Are you saying that the serial cable must also be attached? I have the pins image for the mks sgen_l v2, but do not see any marked as being rx and tx. Would they be located within the AUX1 header(pin socket)?
> > I have attached the image from Makerbase that shows pin numbers and locations.
> 
> Yes, use the AUX1 connector for touch mode.
> EXP1 and EXP2 for Marlin mod

Thank you, this has me a lot closer. I connected the serial connection to the top row of 4 pins in the AUX-1 port with the RESET connector(the one that is not in the molded connection on serial wire) left disconnected on the far right.   I now can get the touchscreen part working.  What I am left with is on the Marlin screen all I get is "LCD12864 Simulator" on the top of the screen, nothing else.   I no longer see the No Printer error message.

A common problem when dealing with software meant for many different brands of hardware is what I call "finger pointing".  I happen to have MKS hardware simply because that was what came in my original FLsun CUBE printer.  I became accustomed to how it connects, and how to configure it using Marlin.  The first time I ran into "finger pointing" was after I had bought a set of TMC2208 drivers from BTT.  Those drivers came with some jumper wires in the box.  I contacted support at the vendor where I bought the drivers and they said I had to contact BTT.  I contacted BTT and they said I had to contact Makerbase.  Makerbase said they didn't know and to contact BTT since they packaged the jumper wires with the drivers.  I eventually found out (on a public forum) that I was very likely going to have to do some soldering and pin movement.  I ended up just buying a set of drivers from MKS that simply plugged in and a simple jumper being moved to enable UART mode.  

I appreciate the help. Trying to deal with manufacturers becomes very difficult with majority of support people not speaking English, and it ends up requiring only talking about one issue at a time.

radek8, 2020-12-31:
Attention! If I remember the sim, the MKS installs connectors EXP1 and EXP2 rotated 180 °
You must therefore remove the cover and turn it 180 °
[image]
[image]

doug-scott, 2020-12-31:
Its bizarre how Makerbase did that on their boards.  I have a SGEN v1 (LPC1769), SGEN_L v1, SGEN_L v2 and a GEN_L v1(original 8bit board that came with printer) and all of them will only work if both cables are reversed.  Really have to wonder why they would shoot themselves in the foot.  They must realize it either drives up their support budget or it reduces their customer base.  I have not tried pulling the "connector guide" off and reversing it, I just used the small pair of angle cutters to cut the tab off of the cable. 

Has anyone else noticed how a lot of the 10 pin ribbon cables are coming with one end backwards?  I bought a couple of longer ribbon cables in Sept 2019 and found one was flipped.  I always watch the red stripe on these cables because that end of the plug goes on the pin 1 side of the socket. At the time I thought it was just crappy QA, now I wonder if they are producing cables with one end reversed.

Now, back to the matter at hand.  First off, thank you very much for the clarification.  One thing to add, on the serial cable they give you with the touchscreen there is one loose terminal.  That one is the reset and just hangs in the air.  At the opposite end of the connector is where the 5v wire is, and connects to the top row in the AUX-1 socket as shown in image above.  As to the idea that this is not really the right place to have this, I would suggest that the information should always be in a place where someone will find it easily.  In my case I tend to tie the firmware to the hardware in issues like this one.  Unfortunately 3d printers have evolved into a component level, right down to the first level of code in the hardware not coming from the manufacturer.  That makes the first level of any kind of tech support out into the freeware world.

The issue with the Marlin portion not loading and just writing "LCD12864" was being caused by the serial cable off by one pin. 
 Once I moved the cable over one pin it worked right away.  All my issues with getting this device to work are solved.  I will come back in the next day or so and will close it.  Thanks again for the information.

radek8, 2020-12-31:
@oldman4U 
If you want to write a notice in the manual for users using MKS boards.
Marlin mod - Rotate connectors EXP1 and EXP2 180 ° (picture above)
Touch mode - connect RS232 to connector AUX1 (upper row). Do not connect the reset. Swap wires RX0 and TX0.

oldman4U, 2020-12-31:
Is this for all MKS boards the same?

Happy new year 😁

radek8, 2020-12-31:
I don't know if all of them, but definitely for
MKS GEN_L Vx.x
MKS SGEN_L Vx.x

doug-scott, 2020-12-31:
> @oldman4U
> If you want to write a notice in the manual for users using MKS boards.
> Marlin mod - Rotate connectors EXP1 and EXP2 180 ° (picture above)
> Touch mode - connect RS232 to connector AUX1 (upper row). Do not connect the reset. Swap wires RX0 and TX0.

Swap wires RX0 and TX0?

radek8, 2020-12-31:
> > @oldman4U
> > If you want to write a notice in the manual for users using MKS boards.
> > Marlin mod - Rotate connectors EXP1 and EXP2 180 ° (picture above)
> > Touch mode - connect RS232 to connector AUX1 (upper row). Do not connect the reset. Swap wires RX0 and TX0.
> 
> Swap wires RX0 and TX0?

Pochopil jsem to tak, že jsi musel zaměnit datové vodiče, aby ti fungoval dotykový režim.
[image]

oldman4U, 2021-01-01:
I decided not to add this information, because I use another board and therefore can not verify this information. It should be added by someone who owns this mainboard.

oldman4U, 2021-01-02:
doug-scott

What is the status of this ticket?

doug-scott, 2021-01-02:
> I decided not to add this information, because I use another board and therefore can not verify this information. It should be added by someone who owns this mainboard.

> doug-scott
> 
> What is the status of this ticket?

Sorry, got tied up with other things and did not get a chance to come back to this.  Am I able to add this information?   How would I go about this?

doug-scott, 2021-01-02:
> Pochopil jsem to tak, že jsi musel zaměnit datové vodiče, aby ti fungoval dotykový režim.
> [image]

Sorry, there appears to be some confusion as to what I meant about serial cable being off by one pin.  I simply meant that when I connected the serial cable to the AUX1 connector I plugged the cable in one pin off.  In other words, I only managed to connect to 3 of the 4 pins.  Only required me to be more careful about plugging the serial cable into the AUX1 connection.

radek8, 2021-01-02:
If you have an EXP1 cable connected, then you only need to connect 2 data wires. The power supply is taken from the EXP1 connector.
if only a serial cable is connected, the connection is in that order

[image]

oldman4U, 2021-01-02:
You can add this using github desktop or VSC and a md editor. You change
the local brach and push a PR to this git repository.

If you send me everything so I only have to copy and paste, I can do it
also.

radek8, 2021-01-02:
I can't edit MD files

oldman4U, 2021-01-02:
I use this one which is easy to use.

https://typora.io

doug-scott, 2021-01-03:
> I use this one which is easy to use.
> 
> https://typora.io

oldman4U,
I can create a document on how to connect the TFT35E3V3 to MKS SGEN and SGEN_L mainboards and attach it to a message here if you like.  You could then cut and paste it to where ever it goes.  Let me know if that works for you please.

oldman4U, 2021-01-03:
Perfect. Thats ok.

doug-scott, 2021-01-03:
ok, I will get that to you later today.

doug-scott, 2021-01-03:
Here is the document.  I used the images from radek8 that were posted here, thank you radek8.   I tried the tool that was suggested here (typora) but could not see how to embed an image into the file.  Did not know where the images were going to reside so I just did it as Word, then saved as PDF.  If you want the word format I have it as well.

[Setup of BTT TFT35.pdf](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5761818/Setup.of.BTT.TFT35.pdf)

jagdpantherx, 2021-01-03:
> Yes, it will work. Well, at least with the Gen-L-v1 that I have. But you will need the nightly version of the BTT-v26 or the stable BTT-v25 version. The stable BTT-v26 will show "no printer [...]

Could you please describe how you managed to get it running with MKS GEN_L V.1. I have the marlin 2.0x firmware and cant get rid of no printer detected error.

radek8, 2021-01-03:
> Could you please describe how you managed to get it running with MKS GEN_L V.1. I have the marlin 2.0x firmware and cant get rid of no printer detected error.

What version of the display do you have?
What version of Marlin are you using?

Submit configuration files to Marlin.
Configuration.h
Configuration_adv.h

oldman4U, 2021-01-03:
Hi doug-scott.

Thank you for the pdf! I updated the ReadMe under my PR #1455, so please check if everything is correct. To add a picture to the md file, you simply copy it to one of your tickets, this one for example, like you would like to upload it. This generates a link, which you can add to the md document, like shown in the attached screenshot.

Please let me know

[image]

oldman4U, 2021-01-04:
Sorry, but I had to load this to a new PR. Please check #1475. In case everything is ok, you can close this ticket whenever you want.

Thank you

doug-scott, 2021-01-04:
> Sorry, but I had to load this to a new PR. Please check #1475. In case everything is ok, you can close this ticket whenever you want.
> 
> Thank you

Looks good to me.  Thanks.
I will leave this ticket open for a bit more for jagdpantherx's use.

oldman4U, 2021-01-04:
Perfect.

Happy printing

doug-scott, 2021-01-12:
Been a week, no activity from jagdpantherx, I will close it now.

closed 2021-01-12 by doug-scott

oldman4U, 2021-01-12:
Thank you. Even the ticket is closed it still can be used.

jagdpantherx, 2021-01-12:
> > > Yes, it will work. Well, at least with the Gen-L-v1 that I have. But you will need the nightly version of the BTT-v26 or the stable BTT-v25 version. The stable BTT-v26 will show "no printer attached."
> > > While the "no printer..." is showing you cannot "talk" to the board. But, if you have the AUX-1 and AUX-2 cables plugged in correctly (the Geln-L-v1 has the back to front) then hold down the BTT button for 3-secs and select "Marlin."
> > > Wait a while (15+ secs) as it needs to boot the Marlin. Once done, you can hold down the BTT button again and go back to Touch and the "no printer" should have gone.
> > > The Gen-L-v1 board does have the 5-pin socket but you only need +vdc, -vdc, Rx and Tx so find them on the v2.0 board. The 5th one is Reset and not needed.
> > 
> > 
> > Could you please describe how you managed to get it running with MKS GEN_L V.1. I have the marlin 2.0x firmware and cant get rid of no printer detected error.
> 
> What version of the display do you have?
> What version of Marlin are you using?
> Did you manage to put the simulated marlin into operation?
> 
> Submit configuration files to Marlin.
> Configuration.h
> Configuration_adv.h

I have uploaded BTT-v25 version and it worked. Thanks alot!

radek8, 2021-01-12:
V.25? Such a history? Don't you want to use it with that?