bigtreetech/BIGTREETECH-TouchScreenFirmware #1461: [Q] How to upload Marlin with Gen-L-v1 board?

FlailAway, 2020-12-30:
How can I update Marlin with the Gen-L-v1 board?

If I go to the Touch "Disconnect" and try to upload Marlin with the Arduino IDE it times out.

Holding down the button in Touch "Disconnect" does nothing.

Currently I have to open the case, unplug the AUX-1 and AUX-2 from the BTT board and plug them into the "Discount LCD..." board.

Then I can use the BTT "Disconnect" and upload Marlin. After which I have to remove the old LCD and re-plug to the BTT.

The "LCD emulator" does not seem to include updating Marlin that I can find. 

How do I do it without having to plug in the old LCD?

Thanks

oldman4U, 2020-12-30:
Hi.

This repository is about the firmware for TFT screens from BTT. How is your request related?

oldman4U, 2020-12-31:
Please help the community and close the ticket once you do not need it anymore. Thank you

FlailAway, 2020-12-31:
> This repository is about the firmware for TFT screens from BTT. How is your request related?

Seriously? It IS related as it IS the firmware preventing me uploading. Please read my OP again.

IF I disconnect the TFT and connect the original RepRapDiscountFullGraphic LCD I can upload Marlin.

With the TFT in Disconnect, I can push the commands like M503 etc. But I CANNOT upload Marlin changes.

This FIRMWARE needs a "Upload Marlin" option.

May I suggest a little less knee-jerk at the mere sight of the word "Marlin" and a little more in-depth reading of posts. This is an issue when using boards that must upload via the Serial connection and the TFT Firmware controls that!

oldman4U, 2020-12-31:
@radek8

Do you know what he means?

radek8, 2020-12-31:
I don't understand the question.
FlailAway, can you better describe your problem?
I understand you have Marlin on Gen_L V1.0
What versions of the TFT display do you have?
Does the marlin simulation mode work on your display?
Do you write that the display is connected via AUX1 and AUX2 ????
Didn't you want to write EXP1 and EXP2?

oldman4U, 2020-12-31:
I believe he complains, that he has to disconnect the TFT physically from the mainboard to be able to update Marlin of the mainboard. Without disconnecting the Arduino IDE does a timeout and disconnecting on the TFT does not work.

This is at least how I understand it - but could be wrong.

radek8, 2020-12-31:
The problem is in the connection of the MKS GEL-L V1.0 board
the same pins are used on the board for USB connection as for TFT connection via AUX-1 connector.
Therefore, if one is connected, the other cannot be connected.
If you connect both at the same time they will argue.
If you do not want to disconnect the TFT if you connect USB, you can map new serial ports and connect the TFT to them via the serial port.
I suggest remapping the end pin Y-MIN -> X-MAX.
This will release Y-MIN and Y-MAX which can work as TX3 and RX3 serial ports.

radek8, 2020-12-31:
location of pins on which the serial port can be enabled:
serial0 - pins D1 and D0
serial1 - pins D18 and D19
serial2 - pins D16 and D17
serial3 - pins D14 and D15

FlailAway, 2021-01-01:
> I believe he complains, that he has to disconnect the TFT physically from the mainboard to be able to update Marlin of the mainboard. Without disconnecting the Arduino IDE does a timeout and disconnecting on the TFT does not work.

YUP, exactly that. Radek8's Port-conflict-response seems to make sense. I will first try lifting the +- 5vdc to the TFT and try that as I can just put in a switch to disable the TFT rather than reassigning ports.

But, shouldn't the TFT Firmware change that reassignment for everyone? Have a 
// Board in use: SKR 1.4 = 1, SKR-1.4-turbo = 2, MKS-S = 3, MKS-Gen-L = 4 ....
"Board: 4"
 etc or whatever in config.ini? The TFT should be working around the obstruction it has created by always grabbing SERIAL_PORT_1.

FlailAway, 2021-01-01:
> Do you write that the display is connected via AUX1 and AUX2 ????
> Didn't you want to write EXP1 and EXP2?

That seems to depend on the board manufacturer. The Reprap-discount... LCD that I have shows "AUX1" and "AUX2." But obviously it means EXP1/EXP2 but I don't get to print the details, on the circuit board. Nit picking does not help. :)

oldman4U, 2021-01-01:
As long as there is no user which can do it the answer  is no.

Personally I think there should be a MKS based repository handling this
kind of requests and not this one here.

radek8, 2021-01-01:
You have no problem with the TFT display.
You need to solve the HW limitations of your MKS board

https://github.com/MarlinFirmware/Marlin/issues/16408

FlailAway, 2021-01-01:
I don't understand what you mean by "no user can do it."

There is nothing wrong with the MKS system, it is the TFT not being able to allow uploading Marlin which is a basic need unless the user has an SKR board and then it is handled by that board.

IF the TFT will **not** work for existing boards other than BTT-32-bit then that should be stated. Right now it is not working with the Gen_L and hope this can be fixed. There are a lot of MKS-Gen-L out there and BTT would be foolish to not try and sell to those users too.

FlailAway, 2021-01-01:
>> You need to solve the HW limitations of your MKS board

I disagree, surely the TFT should check to see if there is a second Serial Port on the board and use that OR find a way to allow the upload. 

We were going to buy these TFT boards for all of our printer farm, but they are mostly running Gen-L boards.

closed 2021-01-01 by FlailAway

radek8, 2021-01-01:
> > > You need to solve the HW limitations of your MKS board
> 
> I disagree, surely the TFT should check to see if there is a second Serial Port on the board and use that OR find a way to allow the upload.
> 
> We were going to buy these TFT boards for all of our printer farm, but they are mostly running Gen-L boards.
> 
> This is enough wasting of my time on this topic.

OMG

radek8, 2021-01-01:
How could the FW in the display solve the problem with the shared port on the Marlin board?
You have to solve the problem in the Marlin configuration and by remapping the RS2320 port to other pins

oldman4U, 2021-01-01:
The software is made by BTT and the software by the open source community, so users like radek8 and me, but with development skills. Sorry that you had to waste so much time. Using an officially supported mainboard would have helped to reduce this time for all of us.

@radek8 do you know if mks support is somewhere mentioned?

radek8, 2021-01-01:
https://github.com/makerbase-mks
???

radek8, 2021-01-01:
description of the same problem

https://github.com/makerbase-mks/MKS-GEN_L/issues/3

oldman4U, 2021-01-01:
Holy moly. And this is a 8 bit board!?

How much time has to be spent to get it to work properly compared to the cost of a new board. And this even does not have to be from BTT....

Thank you radek

radek8, 2021-01-02:
@FlailAway, you have 2 options for software to disconnect the TFT serial port from the Marlin board.
1. Set the - serial_always_on: 0 option to config.ini
after switching to simulated Marlin mode, the TFT should no longer communicate.
2. Click the Disconnect icon

If this doesn't help either, you have to remap the ports on the MKS board.