bigtreetech/BIGTREETECH-TouchScreenFirmware #1464: TFT 70 no touch funktion after update

DeltaLeines, 2020-12-30:
Hello, I bought a BTT TFT70 for my 3D printer. After updating to the latest firmware, the touch function no longer works. What can be the reason? Printer Board SKR1.4 turbo, Marlin 2.0.x.
Greetings Linus

Codel1417, 2020-12-30:
create a file called reset.txt and place it in the sd card

Sebazzz, 2020-12-30:
Same here, TFT50.

oldman4U, 2020-12-30:
Please read this completely and follow the instructions step by step to install and update the TFT:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#connect-the-tft-to-the-mainboard

Sebazzz, 2020-12-30:
Reset.txt helped.

oldman4U, 2020-12-31:
@DeltaLeines Any news?

DeltaLeines, 2020-12-31:
No

oldman4U, 2020-12-31:
Ok

DeltaLeines, 2020-12-31:
Hello, the problem is fixed. The reset with the reset.txt helped. Many thanks for the support.
Best Linus

closed 2020-12-31 by DeltaLeines