bigtreetech/BIGTREETECH-TouchScreenFirmware #1465: Old Marlin interface detects printer, touchscreen doesn't...

HeartlessDarkangel, 2020-12-30:
*I just upgraded my Ender 3 to the SKR Mini E3 V2 and installed everything accordingly and even got the BLTouch working flawlessly, I can use the "old" screen no problem, but when I switch to the touchscreen version it keeps displaying "Printer not found"
I've checked the RS232 cable for continuity and its plugged in properly with the single pin towards the rear of my Ender 3
What could possibly be causing this?
Do I have a defective screen?
How do I perform an update?
TIA!

oldman4U, 2020-12-30:
Hi.

Here is how you perform an update. Read it completely and follow the instructions step by step.

The message means, that the TFT can not see the mainboard, which can have the following reasons:

1. wrong type of serial port defined in Marlin
2. baud rate in Marlin and TFT does not match
3. TFT broken
4. mainboard broken
5. RS232 cable broken or wrong connected

HeartlessDarkangel, 2020-12-31:
> 
> 
> Any news?

Still can't get it to recognize the touchscreen version, ended up just installing the factory screen back...

oldman4U, 2020-12-31:
Do you need further help?

um 16:35:

HeartlessDarkangel, 2020-12-31:
> Do you need further help? HeartlessDarkangel <notifications@github.com> schrieb am Do. 31. Dez. 2020 um 16:35:
> […](#)

I just reflashed the newest firmware and updated fonts and icons and it’s now working 👏🏼
Thank you for your support.

closed 2020-12-31 by HeartlessDarkangel

oldman4U, 2020-12-31:
👍🏻

Happy printing

um 19:27: