bigtreetech/BIGTREETECH-TouchScreenFirmware #1469: [BUG] mesh leveling showing wrong point index with bilinear bed leveling

rightawy, 2021-01-01:
### Description
with bilinear bed leveling  the mesh editor show the point index as I:1 J:65535 when trying to edit a point and the extruder will not move along the X axis only the **I** values will show correctly.

### Steps to reproduce

1. perform ABL with Bilinear bed leveling
2. go to Mesh editor and try to edit a point.

**Expected behavior**
The indexes should be updated to the actual selected point.

**Actual behavior**
the J index is always 65535

### Hardware Variant
TFT35 v2
### TFT Firmware Version & Main Board Firmware details
updated to latest commit on 31-December-2020
Bilinear leveling enabled on Marlin bugfix from 28-December-2020.

digant73, 2021-01-01:
Which is the grid size? and are the grid values correctly displayed in the mesh editor?

guruathwal, 2021-01-09:
I can confirm this bug.

digant73, 2021-01-09:
Strange because I tested all bl types when I developed the editor. Which grid size have you set in marlin? I will retest it

rightawy, 2021-01-11:
4x3 grid size

digant73, 2021-01-12:
I tested a 5x5, 3x4 and 4x3 grid (bilinear bed leveling on Marlin 2.0.7.2, although not the most recent bugfix version) and I had no problem.
Can you provide me the output for the command M420 V1 T1?

In my printer I got the following result:

SENT: M420 V1 T1
READ: Bilinear Leveling Grid:
READ:       0      1      2
READ:  0 +0.088 +0.060 +0.200
READ:  1 +0.093 -0.003 -0.028
READ:  2 +0.063 -0.020 -0.043
READ:  3 +0.075 +0.060 +0.100
READ: 
READ: echo:Bed Leveling OFF
Bed Leveling OFF
READ: echo:Fade Height 10.00
Fade Height 10.00
READ: ok

rightawy, 2021-01-13:
ok I will share the details

rightawy, 2021-01-14:
This is the data received from printer
```
>>> M420 V1 T1
SENDING:M420 V1 T1
Bilinear Leveling Grid:
      0      1      2      3
 0 +0.000 +0.192 +0.230 +0.230
 1 +0.187 +0.320 +0.437 +0.480
 2 +0.155 +0.297 +0.425 +0.375
echo:Bed Leveling ON
```
[image]

digant73, 2021-01-14:
the main difference is the empty line present on my side. Can you confirm that the grid size configured in Marlin FW is 4x3 and not 4x4? See values for GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y in Configuration.h in Marlin FW

rightawy, 2021-01-14:
Yes it is the latest bugfix

digant73, 2021-01-14:
ok, the bug was fixed. After some PRs will be merged, I will open a PR with the fix for that bug plus other features and cleanup

PR #1518 (2021-01-21): Added Home Offset menu + bugfixes and cleanup
**NEW FEATURES:**
1) Added Home Offset menu. Previous ProbeOffset menu is now a generic Z Offset menu (just to not duplicate code) that will invoke Home Offset or Probe Offset API. Both the Home and Probe Offset menus are available on the Bed Leveling menu. Furthermore, Probe Offset menu is available only if probe is enabled (supported) in Marlin fw.
2) Added support to MBL Z offset (G29 S4 Zxx) in Babystep menu. Babystep menu will allow to change/save Z offset based on the following rules (matching sequence):
 *) Probe Offset: If probe is supporte [... cut]
closed 2021-01-21 by bigtreetech