bigtreetech/BIGTREETECH-TouchScreenFirmware #1470: power loss function with skr 1.4 via rs232

sadbass, 2021-01-02:
### Description
the function POWER_LOSS_RECOVERY dont work with tft 35 v 2 via touch mode (rs232) 
setup:
MB skr 1.4 with 
firmware enabled function
   POWER_LOSS_RECOVERY
   HOST_ACTION_COMMANDS
   HOST_PROMPT_SUPPORT
hardware:
no mini ups or other ups
sd with gcode 

screen 
hardware: tft35 v2
no sd 
connected only to MB with rs232 cable
standard firmware (BIQU_TFT35_APP1_V2.0.26.x)

### Steps to reproduce
1. turn on 
2. select print
3. select onboard sd
4. select gcode
5. start print
6. SEE ACTUAL BEHAVIOR

**Expected behavior**
the printer start to print 
if i turn of brutally the power supply and turn on again
appear a prompt (like filament runout sensor) 
showing something like "recovery after power loss founded , continue? yes/no

**Actual behavior**
when i press conferm start print 
the printer not respond anymore (like stuked in a loop)
if i open the terminal (on the tft35) any command i send is not listened by MB 

### Additional Information
sorry for my bad english and tnx to all 
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5760464/firmware.zip)

oldman4U, 2021-01-02:
Not sure if this can work with on board SD when using the Touchscreen, but it works with prints started from TFT based storage.

sadbass, 2021-01-02:
but the tft in not reliable, after some hours of print the printer stop and i don't know why, when i use the sd of the MB don't happen

sadbass, 2021-01-02:
using a lcd12864 doesn't work already?

oldman4U, 2021-01-02:
Well, I only print from TFT and this without any problem. Go to Menu-Settings_Info and send me a picture of what's shown there.

As far as I understand, this function depends on the ability of the TFT to write a kind of notice (Printing.sys) back to the sd card or uDisk at the end of each layer and this is simply not possible for the TFT when printing from on board sd card.

oldman4U, 2021-01-02:
using a lcd12864 doesn't work already?

What do you mean?

The resume print function is inside the firmware of the TFT. See above how it works.

sadbass, 2021-01-02:
[image]

sadbass, 2021-01-02:
And if I leave a SD in the TFT just to store the status? Is possible to integrate this feature?

sadbass, 2021-01-02:
But if I put the gcode I. The TFT SD I must connect the filament runout on the TFT? Or I can leave in the MB?

oldman4U, 2021-01-03:
As far as I know, the status is always written to the storage which is used for printing. But maybe you can program this, I can NOT for sure;-)

I do not use filament runout, so I can not tell you, sorry.

sadbass, 2021-01-03:
does the MB skr1.4 board connected to a screen in emulation mode (ie with ext1 and ext2) not already have the Power loss recovery function?

oldman4U, 2021-01-03:
Maybe. But the serial cable is used in touchscreen mode.

sadbass, 2021-01-03:
I know so for now there is no way to make Power loss recovery work in touch screen mode? if not print from the SD screen?

oldman4U, 2021-01-05:
Thats my conclusion.

sadbass, 2021-01-05:
ok then i will throw away the bigtreetech screen as it has already made me throw away value prints for 20 hours, and i will never buy a bigtreetech product again

oldman4U, 2021-01-05:
Ok. Hope you find the right product for you.

All the best

oldman4U, 2021-01-06:
Please help the community and close the ticket once you do not need it anymore. Thank you

sadbass, 2021-01-06:
why? I still have this problem, when a solution is found I will close it. thanks

oldman4U, 2021-01-06:
Thought you will buy something else so you do not need it anymore.

sadbass, 2021-01-06:
I will buy something else but the problem on these boards remains,

sadbass, 2021-01-08:
if i print from sd on the tft,
the runout filament work? if is attached to the MB? 
because now with this configuration im stucked, if i print from the TFT SD i lose the runout filament detection (2 filament runout)
if i print from the MB i lose the power loss recovery, i've buyed this combo of bigtreetech because fit perfect for my setup 
can anyone help me?

kisslorand, 2021-01-08:
Send M75 at the beginning of the print and you can use your filament runout sensor even if it is connected to the motherboard.
You can set M75 in the start gCode of your slicer or in the start gCode of the TFT (config.ini)

sadbass, 2021-01-08:
Now i try

sadbass, 2021-01-08:
Is workinggggggggg yesssssss tnx a lot

closed 2021-01-08 by sadbass

sadbass, 2021-01-08:
I hope only for the stability of the rs 232