bigtreetech/BIGTREETECH-TouchScreenFirmware #1476: [FR] Save print progress to memory (so printer can be shut down)

gordo3di, 2021-01-05:
**Is your feature request related to a problem? Please describe.**
We've learned that a lot of customer use power loss recovery as a way to save a print to memory and turn off the printer for the night. It would be nice to implement this as a feature instead of relying on PLR.

**Describe the solution you'd like**
A simple button that will create the PLR printing file. Then it will stop the print and maybe home the X Y. This will also allow the printer to clear out the buffer of gcode.

A status screen pops up saying "Print has been saved to memory. You can shutdown the printer" or something.

**Describe alternatives you've considered**
Using the PLR feature, but without the BTT module it will reprint the entire layer. Even with the BTT UPS module it's still buggy and not as reliable as we would like.

Also the user can simply pause a print, but the printer can't be shut down.

**Additional context**
Does anyone think this is useful and worth the effort?

radek8, 2021-01-05:
That would be an interesting feature. But I'm afraid that once the pad cools down, the printed model will no longer stick to the pad enough for printing to work without a problem. For larger models, I would be afraid to turn off printing at night, due to the destruction of the material, if printing is not completed.

oldman4U, 2021-01-05:
Maybe spend some money in 220x drivers. It is hard for me to say if the printer is on or not having it 1m behind me. ;-)

Ciao

gordo3di, 2021-01-05:
> That would be an interesting feature. But I'm afraid that once the pad cools down, the printed model will no longer stick to the pad enough for printing to work without a problem. For larger models, [...]

Yeah I was thinking the same thing. We mostly print on cold acrylic but yep heated beds would be out of the question.

gordo3di, 2021-01-05:
> Maybe spend some money in 220x drivers. It is hard for me to say if the printer is on or not having it 1m behind me. ;-)
> 
> Ciao

We opted for 2130's...

oldman4U, 2021-01-05:
Holy moly.

More quiet is probably not possible then.

oldman4U, 2021-01-09:
Would you like to get this feature request added to the FR list. Do you think it is worth?

gordo3di, 2021-01-11:
> Would you like to get this feature request added to the FR list. Do you think it is worth?

I think it would be something useful and hopefully not too hard to implement. I wonder if when the bed is reheated, how accurate the print would be.

gordo3di, 2021-01-11:
> […](#)

Oh and the 2130's are super silent as well. It's like a whole new printer lol.

oldman4U, 2021-01-11:
If this could work or not also depends on the Z axis stability and how the filament works during cool down and heat up and maybe some other parameters. I can not see this as a function I would use, but again, it is maybe really just me. There also has to be a button for this and the "behaviour" after the printer is turned on again has also to be implemented.

oldman4U, 2021-01-19:
Hi.

Do you want me to add this to the FR list or do you believe you can live without it;-)

gordo3di, 2021-01-19:
I think it's worth adding and I think people (especially schools) would get a use out of it. I was hoping more people might chime in. The more I was thinking about it, it's no different than PLR in regards to the bed cooling down and having to reheat. I'll close the ticket and let you decide ;P

closed 2021-01-19 by gordo3di