bigtreetech/BIGTREETECH-TouchScreenFirmware #1481: Weird behaviour of the TFT 3.5 V3.0

Patrick-81, 2021-01-05:
I connected the TFT 3.5 V3.0 to a BigTreeTech SKR-E3-DIP V1.1 with grey cable and black cable on the TFT plug with reset on up side( driver connection side).
The result is not what I expected. The TFT seems to be frozen on the second screen and don't want to switch to any touch or Marlin mode. And when I see furtively the touch screen I saw a message of a not connected printer.
After some un-succeeded tries, I updated the TFT with a compiled firmware in which I controlled the baudrats and didn't had to set it because this is the default value) as on the card. I didn't had more success.
I checked the card with the old TFT screen and all seems to work fine. 
I had remarked that the screen wasn't well adjusted to the window of the metal support.  As the screen couldn't enter this window there was a cantilever part of it and it was compressed by the screws. I untight the screws and finally all worked fine. I added 2x3mm, half mm thick washers between the metal support and the screen and all now is good.
So to people that could meet same problem the solution is here 8x3mm washers, 2 for each screw.

oldman4U, 2021-01-05:
Hi Patrick.

Before you pressed the "New issue" button, have you read the "Read me before posting a New Issue" right above the button? What you describe is a well known hardware issue, also described in the document mentioned above.

Please check

Thank you

[image]

oldman4U, 2021-01-09:
Any news?

Please let us know

Patrick-81, 2021-01-10:
Hello,
All works fine by now.
Are you from the biqu team ?
Because, if you are, I have a question for you. Does it worth to replace the original fans of the ender 3 by the bigtreetech fans ? Would it make it more silent ?
Regards
Pat

oldman4U, 2021-01-11:
Good to hear.

I am user like you, but can tell you that I replaced almost all the fans with a noctua fan and a 24V to 12V buck converter.

Happy printing

Please close the ticket once you do not need it anymore;-) Thank you

closed 2021-01-11 by Patrick-81