bigtreetech/BIGTREETECH-TouchScreenFirmware #1485: Z motion bug

wtfiAndrewNelson, 2021-01-06:
### Description

Z feed rate is the same as X and Y, meaning that if you use the manual buttons to Move Z it's too fast and the stepper can not move the Z this fast. This is for Slow, Normal, and Fast. I have changed the Slow speed to make it possible to use the Move Z function but that also means that X and Y are now at a crawl.

### Steps to reproduce

1. Press <Menu> button
2. Press <Move> button
3. Press <Z up> or <Z down>

**Expected behavior**

Z should raise and lower at Slow, Normal and Fast speeds (but they are all too high)

**Actual behavior**

Grinding Sound and no motion

### Hardware Variant

BIGTREETECH TFT35-E3 V3.0

### TFT Firmware Version & Main Board Firmware details

BIGTREETECH SKR Mini E3 V2

kisslorand, 2021-01-06:
Z max speed should be set in Marlin:
#define DEFAULT_MAX_FEEDRATE          { X, Y, Z, E0 [, E1[, E2...]] }

wtfiAndrewNelson, 2021-01-06:
The TFT is over riding the marlin features.

kisslorand, 2021-01-06:
Please check:
Menu -> Settings -> Machine -> Parameter -> Max Feed Rate -> /Z

What value do you have there?

wtfiAndrewNelson, 2021-01-06:
12, but it's being ignored by the move button. If I change the slow feedrate in the firmware and re-load it, which is what I have done, then it's fine but like I said that makes the x and y feedrate painfully slow.

kisslorand, 2021-01-06:
That is really strange! Max feedrate cannot be override. This why it is called MAX feedrate.
I just checked, my printer obeys the max feedrate.
I can tell you for sure, the TFT cannot override the MAX feedrate.
Please check what is your Z acceleration, if above 1000 please try to set it at 1000.
If Z acceleration is at 1000 and still have problems with Z move, set the max feedrate to 10, 8... and check again.

wtfiAndrewNelson, 2021-01-06:
Marlin max acceleration is 500

wtfiAndrewNelson, 2021-01-06:
In configuration_adv.h this is my manual move definition:
#define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } 
and it's ignored by the TFT which sets it to slow 1000 normal 3000 fast 5000 and that is superseding the marlin firmware. I have changed my slow to 300 and it works but it takes over all feedrates x y z

kisslorand, 2021-01-06:
I am looking at the code, it is not ignored.

wtfiAndrewNelson, 2021-01-06:
But it is, the TFT is superseding the Marlin, I don't understand how but it is.

kisslorand, 2021-01-06:
OK, let's try a different approach.
What if you send a gCode in the terminal? Let's say `M203 Z10` and than `G0 Z100 F5000`.

wtfiAndrewNelson, 2021-01-06:
I just gave that a try and it ran at 5000ish as it was clearly missing step running that fast. So, M203 was ignored.

wtfiAndrewNelson, 2021-01-06:
I also ran M501 and my feed speed is still the same in the eprom. So both speeds were being ignored and the gcode feedrate superseded it. Unless I am missing something, the TFT commands supersede the Marlin firmware.

kisslorand, 2021-01-06:
It seems to me that Marlin is ignoring the M203 setting. I cannot imagine why.
You sent gCodes, it doesn't matter where you sent them from.
Check the same with Pronterface or Octoprint or whatever else.

wtfiAndrewNelson, 2021-01-06:
Yes it takes it and I get the ok. When I run M503 it reports it as z10. Gcode command supersedes it and it attempts to run at 5000ish buzzing all the way with missed steps. Then when I press the Z up or down button in my new slow speed of 300 it runs fine and at 300. So, once again, the commands from the TFT are apparently taking precedent over the Marlin firmware settings and the volatile memory setting and the eprom settings. So, like I said. The TFT is taking charge and not the Marlin. Now this may be a problem with the BIGTREETECH SKR Mini E3 V2 but it it behaving like it is the TFT overriding everything.

kisslorand, 2021-01-06:
Just test it please with Pronterface or any other external program/device that can send gCodes over USB to Marlin.

wtfiAndrewNelson, 2021-01-07:
Okay. I just tested it by USB and the same thing has happened. So it's not the TFT it's the Marlin limits being ignored by something on the BIGTREETECH SKR Mini E3 V2. I will investigate further. Thank you for your help.

wtfiAndrewNelson, 2021-01-07:
However, that being said. I still need to be able to define Z feedrate to be different than X and Y so that even if there was not a bug in the BIGTREETECH SKR Mini E3 V2 that is allowing gcode feedrates to exceed the maximum feedrate in the Marlin firmware, I would still be able to define a slow, normal and high speed for my Z that is not just the XY feedrate limited to the Z maximum. Maybe this isn't a 'bug' but it's an oversight in the control setup of the TFT

oldman4U, 2021-01-09:
@wtfiAndrewNelson 

Could you please check PR #1487 to see if this solves your issue.

Thank you

wtfiAndrewNelson, 2021-01-13:
That seem to be just the thing.

oldman4U, 2021-01-13:
Sounds good. 

Please help the community and close the ticket once you do not need it anymore. Thank you

PR #1487 (2021-01-15): Separate Z speeds from XY speeds and reduce global variables, wrong home button in TFT70, long file name display during printing from onboardSD & other bugfixes
- Separate Z speeds from XY speeds (#1485 ).
- Reduce global variables.
- Fix long file name display during printing from onboard SD #1245. (This uses 4 bytes of RAM  in comparison to #1488 which uses 256 bytes of RAM).
- Fix Babystep menu opening on pressing the home button after a print is finished on TFT70.
- Removed scroll message from print summary and display "click for Summary" message. 
- Fix print summary message in popup wind [... cut]
closed 2021-01-14 by wtfiAndrewNelson