bigtreetech/BIGTREETECH-TouchScreenFirmware #1486: Getting new errors when compiling latest release

tgioiosa, 2021-01-07:
Just recently I'm getting compiler errors when trying to build BIGTREETECH-TouchScreenFirmware-master release .26.x. that seem to be a problem with USB files. The error list is large so here is an excerpt around the first few errors:

Compiling .pio\build\BIGTREE_TFT35_V3_0 DEBUG UNIQUE\src\src\User\API\CaseLightControl.o
Compiling .pio\build\BIGTREE_TFT35_V3_0 DEBUG UNIQUE\src\src\User\API\Colors.o
Compiling .pio\build\BIGTREE_TFT35_V3_0 DEBUG UNIQUE\src\src\User\API\FanControl.o
Compiling .pio\build\BIGTREE_TFT35_V3_0 DEBUG UNIQUE\src\src\User\API\Gcode\gcode.o
Compiling .pio\build\BIGTREE_TFT35_V3_0 DEBUG UNIQUE\src\src\User\API\Gcode\mygcodefs.o
In file included from TFT\src\User/includes.h:35:0,
                 from TFT\src\User\API\BabystepControl.c:2:
C:\Users\tony\.platformio\packages\framework-stm32cubef2\Middlewares\ST\STM32_USB_Host_Library\Core\Inc/usbh_core.h:28:10: fatal error: usbh_conf.h: No such file or directory
 #include "usbh_conf.h"
          ^~~~~~~~~~~~~
compilation terminated.
In file included from TFT\src\User/includes.h:35:0,
                 from TFT\src\User\API\CaseLightControl.h:9,
                 from TFT\src\User\API\CaseLightControl.c:1:
C:\Users\tony\.platformio\packages\framework-stm32cubef2\Middlewares\ST\STM32_USB_Host_Library\Core\Inc/usbh_core.h:28:10: fatal error: usbh_conf.h: No such file or directory
 #include "usbh_conf.h"
          ^~~~~~~~~~~~~
compilation terminated.
In file included from TFT\src\User/includes.h:35:0,
                 from TFT\src\User\API\Colors.h:9,
                 from TFT\src\User\API\Colors.c:1:
C:\Users\tony\.platformio\packages\framework-stm32cubef2\Middlewares\ST\STM32_USB_Host_Library\Core\Inc/usbh_core.h:28:10: fatal error: usbh_conf.h: No such file or directory
 #include "usbh_conf.h"
          ^~~~~~~~~~~~~
compilation terminated.

If I modify platformio.ini to switch framework=stmcube to framework=cmsis then the errors go away.

Any ideas of what could be wrong?

oldman4U, 2021-01-11:
I guess this is a VSC/PIO issue, because it works perfectly fine on my system. But I have also seen some issues like this before. Google for possible fixes.

[image]

kisslorand, 2021-01-11:
>fatal error: usbh_conf.h: No such file or directory

This should be enough of a hint.

oldman4U, 2021-01-17:
Please help the community and close the ticket once you do not need it anymore. Thank you

tgioiosa, 2021-01-18:
Something is messed up with the stm32cube f2 package that gets automatically installed. So many USB host library errors and some include files that include non-existent .h files.  Instead of going down that rabbit hole I decided to just switch the framework to cmsis and that works with no errors. Hopefully ST will correct the stm32cube USB issues.  I'll close this ticket.

closed 2021-01-18 by tgioiosa