bigtreetech/BIGTREETECH-TouchScreenFirmware #1489: [FR] Increase RAM allocation for BTT TFTs

kisslorand, 2021-01-08:
**Is your feature request related to a problem? Please describe.**
The RAM usage of BTT TFTs are at their very edge.

**Describe the solution you'd like**
@bigtreetech or anybody with the knowledge about the BTT TFTs bootloader RAM usage should make the necessary changes to make the whole RAM available for the BTT TFTs. 

**Describe alternatives you've considered**
Check what I've done for MKS TFT to increase usable RAM to 64KB.

**Additional context**
Some older BTT TFTs are physically limited to 48KB RAM so I guess they are coming to the end of their of upgradeability.
If the rest of BTT TFTs stay at their current 48KB RAM limit (it is a SW limit, not HW) than I guess BTT TFTs are reaching the end of ability for new features being implemented.
I even wonder if the latest 10+ pull requests in the waiting list can be merged without RAM overflow. 
For the moment less than 1KB free RAM is available on BTT TFTs

guruathwal, 2021-01-14:
It does not make sense to discontinue support for devices which have not even completed 2 years of their life cycle and are still being sold. The increased RAM usage is being caused by excessive and inattentive use of global variables for tasks which are either unimportant or can be accomplished without the use of extra variables. Because of this, it is becoming difficult to allocate memory to critical tasks.

kisslorand, 2021-01-14:
So you are basically telling us it is better to ditch support for all the product line WHEN the 48K limit is reached. That really makes a lot of sense...
Reaching the limit of 48K is not an "if" question, it is a "when" question. You can be the mother of all programmers, you will not be able to do every new stuff with local variables. New peripherals, new sensors, new gimmicks appear every day, the demand is high to support them. New features being requested almost every day , the end of 48K is very close.

My request is about to enable the future for the devices that are capable. Why was it necessary again to twist the things I say?
This request is not for me, I already enabled higher RAM for the device I own.

What would be your clever solution WHEN nothing more can be done to keep the RAM under 48K to not "discontinue support for devices which have not even completed 2 years of their life cycle and are still being sold"?

Hukuma1, 2021-01-14:
One potential solution is to do it like Marlin and users choosing during compile which features to include or not.

kisslorand, 2021-01-14:
And keep RAM limited to 48K? So even if some TFT has 128K RAM, the user should select options to fit the firmware into 48K and not use the full potential of the TFT and firmware? Why limit everyone?

...or you mean to raise the limit and give the option selection for those with physical 48K limit?

guruathwal, 2021-01-14:
> So you are basically telling us it is better to ditch support for all the product line WHEN the 48K limit is reached. That really makes a lot of sense...

No, I am not basically telling you this. It is your assumption of my opinion.
Do you think it is better to just ditch the devices with less RAM because the RAM is about to be getting full, without even bothering to try and solve the issue? What is your opinion about the Marlin Firmware still supporting Melzi boards with ATmega1284P? New peripherals, new sensors, new gimmicks appear for marlin also.
even a few weeks ago it was pointed out that MKS TFTs have limited space for storing the parameters (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1359#issuecomment-751525484), but instead suggesting to end the support for these devices, it was ok to remove a feature? does this really make a lot of sense? Does this mean that the end of upgradability is only applicable for the devices with less RAM?
The same type of problem also exists for the SPI flash which is almost 96% full and icons for new peripherals, new sensors, new gimmicks, new features are being added almost every other week. With this, every device with 8MB of SPI will have their end of upgradability in no time.
What would be THE solution WHEN this happens?

>You can be the mother of all programmers

I am sorry, I am not able to understand, what do you mean by this?

kisslorand, 2021-01-14:
> > So you are basically telling us it is better to ditch support for all the product line WHEN the 48K limit is reached. That really makes a lot of sense...
> 
> No, I am not basically telling you this. It is your assumption of my opinion.

So, you agree to move on to more available RAM with the TFTs that can have more than 48K WHEN the time comes?

> Do you think it is better to just ditch the devices with less RAM because the RAM is about to be getting full, without even bothering to try and solve the issue? 

Absolutely not! I think all the TFTs should be maintained as long at possible with responsible programming. I said it once and will say it again, you did and do a great job taking care of other's carelessness or lack experience regarding RAM usage.
 
>What is your opinion about the Marlin Firmware still supporting Melzi boards with ATmega1284P? New peripherals, new sensors, new gimmicks appear for marlin also.

Great job! However it comes with a caveat, not all options (reasonable ones) can be activated.

> even a few weeks ago it was pointed out that MKS TFTs have limited space for storing the parameters ([#1359 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1359#issuecomment-751525484)), but instead suggesting to end the support for these devices, it was ok to remove a feature?

The space for parameters is the same for all TFTs, 0x0800. (It starts from 0x08040000 - 0x800, in that comment I typed mistakenly 0x08004000 and 0x08007000)

> does this really make a lot of sense? Does this mean that the end of upgradability is only applicable for the devices with less RAM?

Yes, even thou it's not the only limiting factor.

> The same type of problem also exists for the SPI flash which is almost 96% full and icons for new peripherals, new sensors, new gimmicks, new features are being added almost every other week. With this, every device with 8MB of SPI will have their end of upgradability in no time.

Yup, that's the other space limiting factor.

> What would be THE solution WHEN this happens?

I don't know.

> > You can be the mother of all programmers
> 
> I am sorry, I am not able to understand, what do you mean by this?

It is a figure of speech. Sorry, my bad. I should have written "One can be the mother...". It wasn't aimed at you.

guruathwal, 2021-01-15:
> So, you agree to move on to more available RAM with the TFTs that can have more than 48K WHEN the time comes?

Not that either. My advice is to look into the issue and solve it. By solving the issue the compatibility of devices with 48K RAM will keep on going and there will be no harm to devices with high RAM. What good will be a decision taken without putting effort to solve the cause? Look at the PRs merged during last 3-4 months. Just see the increase in global variables. For instance, more than 180 Bytes of RAM is required for displaying a scrolling message after a print is finished. Is it not easier and faster to just click and read a long message instead of reading the message letter by letter.
And with "solving the issue" I do not suggest to just start removing the features. The code needs to be optimized to remove any dependency of global variables where ever it is possible, like in #1487, #1384 and #1321.

> > What is your opinion about the Marlin Firmware still supporting Melzi boards with ATmega1284P? New peripherals, new sensors, new gimmicks appear for marlin also.
> 
> Great job! However it comes with a caveat, not all options (reasonable ones) can be activated.

If you compare it to newer 32bit boards from an average user's perspective there will be no caveat (apart from the fancy LCD support and extra hardware support which are not supported due to many other hardware issues, not just RAM) and even with the caveat, there will be more features in Marlin compared to this.

> > even a few weeks ago it was pointed out that MKS TFTs have limited space for storing the parameters ([#1359 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1359#issuecomment-751525484)), but instead suggesting to end the support for these devices, it was ok to remove a feature?
> 
> The space for parameters is the same for all TFTs, 0x0800. (from 0x0803F800 to 0x08040000), in that comment I typed mistakenly 0x08004000 and 0x08007000)
> 
> > does this really make a lot of sense? Does this mean that the end of upgradability is only applicable for the devices with less RAM?
> 
> Yes, even thou it's not the only limiting factor.

In this case, your opinion is looking biased. On one hand, for maintaining compatibility of MKS TFTs, a feature is removed which took only 2 bytes, and on the other hand, in the same PR a feature is added which occupied 180 bytes of RAM without weighing the effect on devices with 48K RAM and suggesting to end support for these devices.

> Yup, that's the other space limiting factor.
>
>
> I don't know.

If the support for devices is ended then we can maintain support for TFT70 which have 32MB flash or else support for all the product line will end. Or is there any other consideration we missed?

ETE-Design, 2021-01-16:
@kisslorand Mabye @Hukuma1 think it is okey to raise RAM, but still also support TFT with less RAM... Mabye something like Marlin does now, when you choose the board with less ram you have to choose between features you like to use... In that way people with TFT's with 48KB RAM have to choose what features they like to have enabled, and people with 128K RAM TFT's can have it all enabled...

carryreynolds, 2021-01-26:
It feels pity to see MKS users deciding the fate of BTT TFTs. The objective of this repository is now changing to maintaining compatibility for MKS TFT instead of developing for BTT's TFT. @bigtreetech why is this firmware being made compatible with MKS TFT? MKS have its own firmware, why don't they use that.

kisslorand, 2021-01-26:
I can confirm! I just had a warning on my BTT TFT35 v3.0 screen...

[image]

carryreynolds, 2021-03-02:
[image]

kisslorand, 2021-03-07:
Thank you Alan for PR #1562.

closed 2021-03-07 by kisslorand