bigtreetech/BIGTREETECH-TouchScreenFirmware #1493: Cant read SD Onboard card (TFT35 / SKR TURBO

Grex3D, 2021-01-09:
HI,

Im trying to print with the sd card onboard but nothing work.

I use sdsuuport on/off
SD lcd connection on/off
etc

Nothing works. When i go in print > onboard SD the screen beep 2 or 3 time and return to print menu.

Someone can help me ?

blueeagle69, 2021-01-09:
Hi.

Check Configuration_adv.h in the Marlin firmware to see if the following is set.
**#define SDCARD_CONNECTION ONBOARD**

Grex3D, 2021-01-09:
> Hi.
> 
> Check Configuration_adv.h in the Marlin firmware to see if the following is set.
> **#define SDCARD_CONNECTION ONBOARD**

Yep but in the last version of Marlin bugfix i cant find this line.

blueeagle69, 2021-01-09:
Strange, I just checked the latest Marlin Bugfix code and the setting is there.
Line 1396.

Grex3D, 2021-01-09:
OK !!!! i understand now :)

I have to replace  LCD by ONBOARD !

That why i cant find #define SDCARD_CONNECTION ONBOARD :D

Im gonna try :)

blueeagle69, 2021-01-09:
No probs.
Hope it sorts it.

Grex3D, 2021-01-09:
Perfect its work
(And the TFT35 SD work too)
 thank you :)

closed 2021-01-09 by Grex3D