bigtreetech/BIGTREETECH-TouchScreenFirmware #1494: [FR] Different Distance X/Y - Bed Level Knop

ETE-Design, 2021-01-09:
A lot of printers dosen't have the level knop under the bed at the exact same distance from both X and Y axis, so would be great if you could set both distances at different messurements.

oldman4U, 2021-01-09:
What for those using only 3 knobs?
And where can this be defined today in config.ini or ??

As far as I understand, you can define the xy distance for the manual levelling, but this is independent of the position of the screw.

ETE-Design, 2021-01-09:
@oldman4U Or for people with 4 knobs not placed at the same distance from X and Y... Normal you level your bed just over the level knob... Today you set the distance in this line `level_edge_distance:20` and then X and Y is equal... On a lot of printers the knop is not in the same position from X and Y...

oldman4U, 2021-01-10:
But this distance is completely unrelated to the knob distance. For me this makes more sense in case the bed is not a square, but even then, the difference will be ignorable I guess.

Let's see how many other users believe this could be a useful function.

closed 2021-03-18 by stale[bot]