bigtreetech/BIGTREETECH-TouchScreenFirmware #1497: [BUG] Filament runout Bug to resume

Bastlwastl84, 2021-01-09:
Marlin Mode: starting Print
Long touching TFT. 
Switched to Touchmode
Filament Runout happens
At infoscreen:
[image]
Pressing play -button doesn't take effect. Short showing pause -button, then resume-button.
Cant resume.
Can't switch back to marlin mode.

Opening extruder Menu make the Tft freeze

Skr Mini E3 V2
Marlin Bugfix 2.0.8.x from Yesterday
TFT 35 E3 V3
Firmware downloaded yesterday
Config at next posting

Bastlwastl84, 2021-01-09:
[2021-01-08 skr mini v2 marlin + tft35e3v3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5791706/2021-01-08.skr.mini.v2.marlin.%2B.tft35e3v3.zip)

3dfixer, 2021-01-11:
I've been pulling my hair out on this for days. got a sensor on e0det plug, which is where it should go.
this works fine, doing a m119 sends back open or triggered depending if theres filament.
the marlin side of the tft see this and does a filament change, then resumes where it left off. this is all done in config.h in marlin firmware.
All seems ok then??
However, the fancy icon side of the tft is looking for a sensor connected to the tft itself. As there is not one, it will trigger a false runout all the time. Doing a m119 reads the sensor on e0det, not a sensor on the tft.
Perhaps BIGTREETECH will look at sorting this out???!!

also, whatever side you start a print from, its best to stay there. Strange things happen when you swap.

hope this helps people

kisslorand, 2021-01-11:
If you have the filament runout sensor connected to the motherboard, try flipping the `fil_runout_inverting:` parameter in `config.ini`. I cannot remember which one but one of the settings will give you false alarm.

kisslorand, 2021-01-12:
Please try with M75 before print starts. I think this was what happened to me also and made me send M75 before every print.

Bastlwastl84, 2021-01-12:
At beginning of the startcode once?

kisslorand, 2021-01-12:
Yes, that is enough.
You can also include it into the start gCode in config.ini.

gerthalberg, 2021-01-12:
Giving Marlin the M75 command in the start gcode to enable filement runout triggering has to do with letting Marlin know that there's a print job running. I've just got my filament runout to work on my SKR 1.4 board in Marlin doing just that (I.e. I dis this #define PRINTJOB_TIMER_AUTOSTART). So the M75 is solely for getting Marlin to react but the question is - does the TFT react to an unsolicited Marlin event that's not generated by first sending a command from the TFT?

kisslorand, 2021-01-12:
>  does the TFT react to an unsolicited Marlin event that's not generated by first sending a command from the TFT?

Yes. TFT reacts to what it gets on its serial port from Marlin, regardless of whoever triggered that.

guruathwal, 2021-01-13:
The messages displayed on the screen are coming from the Mainboard. 

The Runout sensor setting provided in **config.ini** (`fil_runout_inverting:` & `fil_runout:0`) are for connecting the sensor to the TFT controller.  If the runout sensor is connected to the mainboard, then you need to check the documentation for marlin firmware.

@Bastlwastl84 in the files you provided. the `HOST_ACTION_COMMANDS` is not enabled. You have not provided the configuration you used for the TFT controller. You need to check if you have enabled `serial_always_on` in for you TFT controller. It is not recommended to change the mode during printing. if `serial_always_on` was enabled in your TFT, it would have prevented switching to touch mode during printing.

Bastlwastl84, 2021-01-14:
> The messages displayed on the screen are coming from the Mainboard.
>
> The Runout sensor setting provided in **config.ini** (`fil_runout_inverting:` & `fil_runout:0`) are for connecting the sensor to [...]

Now i had enabled, but now new bug ( marlin had problem with powerless recovery. Print doenst start and an error comes... Need to figure out where ist that problem, than i could check for more. Sending gcode with Esp01 makes problems with freezing tft, too (reported as problems with serial Communication)

Bastlwastl84, 2021-01-14:
Setted the Pin of powerless to -1, cause of not possible to start a print without message, that a powerfailure happened.
Activated HOST_ACTION_COMMANDS
[2021-01-14 skr mini E3 v2 marlin bugfix + tft35e3v3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5815290/2021-01-14.skr.mini.E3.v2.marlin.bugfix.%2B.tft35e3v3.zip)
Here is what happened:
https://youtu.be/O32BzSgElAQ

guruathwal, 2021-01-16:
@Bastlwastl84 That is because marlin is not responding.

closed 2021-03-24 by stale[bot]