bigtreetech/BIGTREETECH-TouchScreenFirmware #1498: [BUG] Touch is not working after update to last firmware TFT24 V1.1

mrigne, 2021-01-09:
### Description

Touch is not working after update to last firmware TFT24 V1.1

### Steps to reproduce

1. Download firmware, language pack, fonts and icons to SD card
2. Insert card into display
3. Power On printer

**Expected behavior**
All updated without errors
After screen booted - touch screen is working

**Actual behavior**
Marlin mode is ok, but in touch mode screen does not react to tuches

### Hardware Variant
BTT TFT24 V1.1
### TFT Firmware Version & Main Board Firmware details
[BIGTREE_TFT24_V1.1.26.x.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT24_V1.1.26.x.bin)
Main Board - SKR Pro 1.2, Marlin 2.0.7.2

### Additional Information

oldman4U, 2021-01-09:
Please follow the update procedure described here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#update-tft-firmware which also includes the config.ini.

There you will also find a Troubleshooting section which eventually could also help

Let us know

INeDiA, 2021-01-09:
Exact same behaviour and config here.
The touch screen reads the input (when in idle and it is touched, the brightness goes back to normal) but the icons can't be selected.

oldman4U, 2021-01-09:
And Troubleshooting?

mrigne, 2021-01-09:
@oldman4U I have tried to create reset.txt inn root of SD, but nothing happened.
According to source code if file exists - touchscreen calibration should be started and reset.txt should be renamed to reset.txt.DONE, but screen just booted and nothing happened. 
I will try to build my own binary from sources but bus exists

oldman4U, 2021-01-09:
Yes. Touchscreen should be presented.

TFT firmware works on my TFT24v1.1 with SKR Pro v1.1, same Marlin version.

Remove all cables from TFT, cut a USB cable and connect red and black wire to 5V and GND of the TFT. Power it up and try to reset again.

mrigne, 2021-01-09:
@oldman4U just try to create reset.txt in root
I have changed in boot.h
#define TFT_RESET_FILE "0:reset.txt" > #define TFT_RESET_FILE "reset.txt" 
(similar to firmware filename which does not have 0: prefix) and rebuilt firmware and updated
After this I have created reset.txt in root and all was reset successfully.
So looks like there is bug exists - need to change in boot.h next strings:
#define ADMIN_MODE_FILE "0:admin.txt" -> #define ADMIN_MODE_FILE "admin.txt"
#define TFT_RESET_FILE "0:reset.txt" > #define TFT_RESET_FILE "reset.txt"

@INeDiA if you interested - firmware whit only one string changed:
[BIGTREE_TFT24_V1.1.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5791953/BIGTREE_TFT24_V1.1.26.x.zip)

I have also uploaded config.ini and language file with this file because invalid value(s) error shown.
Now all is working fine

oldman4U, 2021-01-09:
Thats great, thank you. Saw your PR...

The question is still, why you had to reset the TFT after FW update?!

mrigne, 2021-01-09:
For me right after fw update touchscreen was not reacting to touches. So I was unable to use touch mode - only marlin mode was usable using encoder

INeDiA, 2021-01-10:
> @INeDiA if you interested - firmware whit only one string changed:
> [BIGTREE_TFT24_V1.1.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5791953/BIGTREE_TFT24_V1.1.26.x.zip)

@mrigne Thanks, but manually resetting solved the issue for me :)

oldman4U, 2021-01-10:
Seems it works for all of you now.

Please close the ticket once you do not need it anymore.

mrigne, 2021-01-10:
Sure, thank you

closed 2021-01-10 by mrigne

davbcr, 2021-02-09:
I've the very same problem too. After update FW to V.1.1.26.x touch doesn't work anymore. I tried also the INeDiA repo files but it doesn't work either..... what can i do? i'm a newbie guys have patience.

oldman4U, 2021-02-09:
You can read the ReadMe incl the trouble section part which explains how to reset the tft.

Happy printing

woodcrafter10, 2021-02-27:
Why in the world would you have to do a reset on a firmware update, shouldn't that, if needed, already have been in the updated firmware to do so one time on the update?  A week chasing down this crap, poor setup indeed.  Still can't get the printer and tft to talk to each other "No printer attached"

oldman4U, 2021-02-27:
Every user can fiddle around and waste time or read the Readme and do what is clearly described there.

It is a free world.

No printer attached means that the TFT can not see the mainboard on the other side through the RS232 connection.

kisslorand, 2021-02-27:
Why in the world would someone spend so much time "chasing down this crap" instead of 5 minutes to read the Readme???

woodcrafter10, 2021-02-28:
Well, the real problem turned out to be a corrupted file on the SD card.
once I did a *full*  *format,* not the *quick format*, and moved the proper
files over everything worked.  The readme had no help there that I remember
seeing.  So making sure you have a clean SD card is imperative.  No Printer
attached can mean a lot of things besides the UART not functioning, it
could be the install is corrupted and can't communicate in any mode.