bigtreetech/BIGTREETECH-TouchScreenFirmware #1500: [Q] How to manually level the bed using the touch screen?

INeDiA, 2021-01-10:
Hello, I'm using a TFT24 v1.1 on a SKR mini v1.2, both updated with the latest firmwares.
I own a Prusa clone, so the bed moves on Y and the noozle on XZ.

I'm trying to level my bed manually since I don't own a bltouch.
I select Movement, Leveling, then one of the points. When the noozle reaches the surface I am supposed to finetune Z, but how can I do that?
On Marlin interface you just use the knob, then push to confirm.

I couldn't find any video/manual available online, sorry for asking this here.
Thanks for help!

oldman4U, 2021-01-11:
Manually means changing the position of the bed manually.

kisslorand, 2021-01-11:
In each point adjust the bed's height by the adjusting screws under the bed. Do it using a sheet of paper, credit card, a feeler gauge or whatever else suits you. Adjust them so you'll have the same distance (more or less) between the nozzle and the bed.
After that adjust the Z offset.
Here's a link after a quick search on Google explaining in details: https://all3dp.com/2/3d-printer-bed-leveling-step-by-step-tutorial/

INeDiA, 2021-01-11:
I just tought this was the same as the Marlin bed leveling, where you fix Z for each corner.
Thanks everyone for your help!

closed 2021-01-11 by INeDiA